yamcs/yamcs #609: Using Yamcs APIs internally from a plugin

merose, 2021-09-22:
I would like to call Yamcs functionality from a plugin. For example, have a plugin send a command automatically. It's unclear at what level the plugin should invoke the command.

One possibility is to look at what CommandsApi.issueCommand() does, and copy the code that seems appropriate, probably the portion that creates a PreparedCommand, then get the processor, the commanding manager from the processor, and send the command.

Another possibility is to use ProcessingClient.issueCommand(). This seems a bit more straightforward in that it matches the HTTP API, but it requires passing authentication, and I'm not sure how that would work from within Yamcs.

Which of these two approaches would be preferred, to access Yamcs functionality from within a plugin?

fqqb, 2021-09-23:
Both are valid. I generally prefer first possibility because it is more direct (no HTTP involved, and indeed no auth required). But more verbose for sure (a sign that perhaps some of the logic in the API should be moved to general services).

There is currently no special support for using the java client from within a plugin. Auth and connectivity need to be handled same as if the client were used in a different program.

merose, 2022-06-20:
Closing this because it is a query and has been answered.

closed 2022-06-20 by merose