yamcs/yamcs #610: Javascript padStart not a function

nmaas87, 2021-09-23:
Dear all,

I tried to use some simple calculations within the Algortihm part of my MDB / Yamcs Server "Algorithms" using Javascript.
At one point I need to just "pad" an integer with zeros for the miliseconds part and just tried something tried and proven:

````
ms_mt = ms_mt.toString().padStart(3, '0');
````

However, this resulted in:
````
TypeError: ms_mt.toString().padStart is not a function in <redactedAlogo> at line number 43: ms_mt = ms_mt.toString().padStart(3, '0'); 
````

Could it be that padStart and the overall functionality is not yet included in the Javascript part of Yamcs?
It should be part of ES2017, so quite some years already...
https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/String/padStart

Is that maybe still missing and could a newer engine than added to Yamcs?

Thanks a lot!

fqqb, 2021-09-23:
The JavaScript implementation embedded in JDK 8 or 11 is called Nashorn, and is ECMAScript 5.1, with a little bit of ECMAScript 6. So yes padStart would be too new. JavaScript has changed a lot in the past few years, which is one of the reasons Oracle stopped bundling Nashorn as of Java 15.

More detail on Nashorn is here: https://github.com/openjdk/nashorn

nmaas87, 2021-09-23:
Thanks for the headsup :)!

closed 2021-09-23 by nmaas87