yamcs/yamcs #615: Packet Viewer Parameter search does not seem to work

nmaas87, 2021-10-01:
Dear all,

I tried to search for a parameter with a known value, to when it turned to this value (e.g. eventMode == 102) - however, whether I try this on Windows or Linux (WSLv2 Debian), the cmdline shows

````
Provide .. /<REDACTED>/<REDCATED>/eventMode
````

but nothing happens, even though there are packets with eventMode == 102 are within the loaded telemetry data.

I am using the latest packet-viewer version, but cannot tell when this error was introduced - as I am just using it from time to time / debugging.

[image]
[image]

fqqb, 2022-10-12:
Bit late here, but do you recall what type of parameter this is?
Numeric comparison would not work for an enumeration parameter, but seems fine for integer as far as I can tell.

nmaas87, 2022-10-12:
Hey @fqqb - it was an integer / uint8. I just had a look at it with the latest packet-viewer and I see the same issues, I cannot find the searched parameter. However, I found something new: Each packet starts with a normal CCSDS header, meaning the packet version is the first item. And I can search for the first items, but at some point I cannot search anymore for items, even if they are the same type. Also, it seems to "glitch" sometimes and does allow searches for one parameter - and on another try not. There is definitly something wrong with the search in the packet-viewer, its a good tool for "quick look" - but it really needs the capability to have a reliable search. Thank you for looking into it and coming back to me :)