yamcs/yamcs #616: Arrays For Commands

lorenzo-gomez-windhover, 2021-10-01:
Hi there,

Hope you are all doing well.

Are there any plans for adding support for `ArrayArgumentType`?

Thanks
Lorenzo

fqqb, 2022-10-04:
As of v5.7.7, single-dimensional array arguments are supported.
* XTCE StartingIndex has to be 0.
* XTCE EndingIndex can be specified using either FixedValue or DynamicValue (argument or parameter)

Keeping open because multi-dimensional array arguments will not yet work.

snthzm, 2024-09-16:
@fqqb any updates on multi-dimensional array supports ?

fqqb, 2024-09-17:
Since this issue, the UI should work for multi-dimensional array arguments (or it did at some point).
But the backend is for @xpromache to have a look at. From what I remember, there's a pending issue relating to determining the correct dimension lengths.

In practice, we don't really see this kind of arguments in our projects, so it's never been prioritized.

snthzm, 2024-09-17:
I was wondering because ArrayArgumentType taking another ArrayArgumentType as an arrayTypeRef doesn't seems to be supported by the Commanding manager (cannot extract the data I put for the inner array element) with the following error message : 
`RouteHandler 16: Responding '400 Bad Request': Cannot assign value to packetStoreSequence: bad size for array of type TC-XX-X-TCNAME__packetStoreSequence__packetStores__el; expected size : 1, actual size: 0`
Have you noticed that ?

closed 2025-03-04 by xpromache