yamcs/yamcs #618: Realtime filler does not handle packets out of order

merose, 2021-10-02:
Symptom: With the realtime filler enabled, sending two packets where the 2nd has a timestamp 2 seconds before the first causes the 2nd packet not to be added to the parameter archive.

To reproduce:

First, download this tar-ball, which modifies the quickstart project:
[old-packets-with-realtime.tar.gz](https://github.com/yamcs/yamcs/files/7270704/old-packets-with-realtime.tar.gz)

Then,

1. Clone the yamcs/quickstart project: `git clone git@github.com:yamcs/quickstart.git`
2. cd to the quickstart project directory: `cd quickstart`
3. Unpack the downloaded tar-ball: `tar xf <path>/old-packets-with-realtime.tar.gz`
4. Build quickstart: `mvn package`
5. Run Yamcs: `target/bundle-tmp/bin/yamcs`

Then, in another terminal window, send two packets using an included script, with the 2nd packet 2 seconds older than the first, show the archived values of a packet parameter to see that only one value is archived, and see the error logged by the RealtimeArchiveFiller:

```
$ # Send two packets with the 2nd 2000 milliseconds older than the first.
$ python3 send_packets.py --second-packet-offset -2000 1 2
Sent two packets with times 1) 2021-10-02 01:24:48.893752+00:00 2) offset by -2000 milliseconds
$ python3 show_archived_values.py # Note that only one value is shown, not two.
2021-10-01T18:24:48.893000-07:00: value=1
$ tail -n 15 target/bundle-tmp/log/yamcs-server.log.0 # See the warning about dropping the 2nd packet.
parameterFqn: "/myproject/packetSequenceCount"
parameterType: 196610

Oct 01 18:24:48.899 _global [27] org.yamcs.yarch.rocksdb.Tablespace [FINE] Adding new metadata record tbsIndex: 41
type: PARCHIVE_DATA
parameterFqn: "/myproject/value"
parameterType: 131074

Oct 01 18:24:48.995 myproject [27] org.yamcs.parameterarchive.RealtimeArchiveFiller [WARNING] Dropping old data with timestamp 2021-10-02T01:24:46.893Z (minimum allowed is 2021-10-02T01:24:46.893Z).Unsorted data received in the realtime filler? Consider using a backfiller instead
Oct 01 18:24:55.289 _global [26] org.yamcs.http.HttpServer [FINE] [id: 0xac1ce9bc, L:/0:0:0:0:0:0:0:0:8090] READ: [id: 0xbfab39aa, L:/0:0:0:0:0:0:0:1:8090 - R:/0:0:0:0:0:0:0:1:51138]
Oct 01 18:24:55.290 _global [26] org.yamcs.http.HttpServer [FINE] [id: 0xac1ce9bc, L:/0:0:0:0:0:0:0:0:8090] READ COMPLETE
Oct 01 18:24:55.316 _global [30] org.yamcs.http.HttpRequestHandler [FINE] bfab39aa GET /api/archive/myproject/parameters/myproject/value?source=ParameterArchive&order=asc&limit=2&processor=realtime
Oct 01 18:24:55.321 _global [30] org.yamcs.http.HttpRequestHandler [FINE] 1: Routing GET /api/archive/myproject/parameters/myproject/value?source=ParameterArchive&order=asc&limit=2&processor=realtime
Oct 01 18:24:55.333 _global [30] org.yamcs.http.HttpRequestHandler [INFO] bfab39aa GET /api/archive/myproject/parameters/myproject/value?source=ParameterArchive&order=asc&limit=2&processor=realtime 200
Oct 01 18:24:55.334 _global [30] org.yamcs.http.Context[c1] [FINE] API call finished successfully, transferred bytes: 0
$
```

It appears that the logic in RealtimeArchiveFiller is wrong (lines 229–241 in RealTimeArchiveFiller.java):

~~~
                if (t < segHead.getSegmentStart() - pastJumpThreshold) {
                    log.warn("Time jumped in the past; current timestamp: {}, new timestamp: {}. Flushing old data.",
                            TimeEncoding.toString(segHead.getSegmentStart()), TimeEncoding.toString(t));
                    flush();
                } else if (t < segHead.getSegmentStart()) {
                    log.warn("Dropping old data with timestamp {} (minimum allowed is {})."
                            + "Unsorted data received in the realtime filler? Consider using a backfiller instead",
                            TimeEncoding.toString(t),
                            TimeEncoding.toString(segments[head].getSegmentStart() - sortingThreshold));
                    return;
                } else {
                    writeToArchive(t);
                }
~~~

The cases are, in order: 1) packet times older than now minus pastJumpThreshold, 2) old packet times but not that old, and 3) newer packet times.

But the code inside the cases looks like they belong to 2, 3, 1, in that order. In other words, I think the correct code may be:

~~~
                if (t < segHead.getSegmentStart() - pastJumpThreshold) {
                    log.warn("Dropping old data with timestamp {} (minimum allowed is {})."
                            + "Unsorted data received in the realtime filler? Consider using a backfiller instead",
                            TimeEncoding.toString(t),
                            TimeEncoding.toString(segments[head].getSegmentStart() - pastJumpThreshold));
                    return;
                } else if (t < segHead.getSegmentStart()) {
                    log.warn("Time jumped in the past; current timestamp: {}, new timestamp: {}. Flushing old data.",
                            TimeEncoding.toString(segHead.getSegmentStart()), TimeEncoding.toString(t));
                    flush();
                } else {
                    writeToArchive(t);
                }

~~~

merose, 2021-10-02:
Found a workaround, but not sure of the implications. Change the realtime filler configuration as follows:

~~~
  - class: org.yamcs.parameterarchive.ParameterArchive
    args:
      realtimeFiller:
        enabled: true
        sortingThreshold: 600000
        pastJumpThreshold: 0
~~~

This causes _all_ old packets to fall into case 2 in the current code. The sorting threshold is increased to a large value so that new packets up to 5 minutes old will fall into the current segment.

xpromache, 2021-10-09:
I think the second if is wrong, instead of 
> else if (t < segHead.getSegmentStart()) {

it should be
> else if (t < segHead.getSegmentStart() - sortingThreshold) {

Then it would also match the error message.

commit 4f47db78 (2021-10-09): Fix realtime parameter archive filler

The sorting threshold was not working properly.

Fixes #618