yamcs/yamcs #620: Should support initialValue in AggregateArgumentType definitions

merose, 2021-10-10:
It would be convenient if the initialValue specification in AggregateArgumentType definitions were used when sending a command or adding commands to a command stack.

fqqb, 2022-09-06:
`<AggregateArgumentType />` itself does not have initialValue support (no such field in XTCE, some other types do have it).
But I added a commit to #714 which should allow to specify what is requested.

Yamcs will then consider aggregate initial values in the following priority order:
1. `initialValue` attribute specified on the aggregate `<Argument />`.
2. `initialValue` attribute specfied on a specific member of the aggregate: `<Member />`
3. `initialValue` attribute specified on a the type of a specific member of the aggregate: `<IntegerArgumentType />`, etc

In case of (1), the aggregate value needs to be encoded in JSON (use \&quot; to escape ") format. Note that this may be Yamcs-only behavior, because XTCE does not appear to define how the `initialValue` attribute should be interpreted for aggregates (if at all).

In principle, the `initialValue` specified through (1) needs to be complete (that is: all members present). Although if a member is missing that does have `initialValue` set using (2) or (3), everything is merged together, and accepted by Yamcs.

By using (2) and (3) *without* (1), you could have some members of the aggregate initialized, and others not.

Here is an example for (1):

```xml
<ArgumentList>
	<Argument argumentTypeRef="Config_Item_Type"
                  name="CFGITEM" 
                  initialValue="{&quot;ADDRESS&quot;: 44, &quot;VALUE32&quot;: 6}" />
</ArgumentList>
```

fqqb, 2022-09-30:
On master

closed 2022-09-30 by fqqb