yamcs/yamcs #630: Should either not truncate command arguments in command history or have an option to disable it

merose, 2021-11-03:
Currently the command arguments assignments are truncated when shown in the command history, followed by an ellipsis (...). For commands with long argument lists, especially those generally submitted by a script, this means it's sometimes impossible to tell what a command is doing unless you can parse the arguments from the hex display.

Either don't truncate command argument assignments in the GUI or make it an option that can be disabled.

fqqb, 2021-11-22:
I don't see any ellipsis in command history?

What's true, is that arguments which are not explicitly set by the commanding user are not visible (because they are inheritance assignments, or use their default value). Is that maybe what you're referring to?

merose, 2021-11-22:
Here are two examples. I've issued a command which has 5 arguments, 4 of which are identical aggregates argument types with 13 members each. In both the "Send Command" page and the "Command Detail" panel, only 5 of the 13 members are listed. None of the arguments that are omitted are inheritance assignments.

Send Command page:
[image]

Command Detail panel in command history:
[image]

fqqb, 2022-08-10:
Should be better now on master.
I made a component that allows to interactively expand nested array or aggregate values starting at the top level.

closed 2022-08-10 by fqqb