yamcs/yamcs #633: Parameter data with generation time within the same millisecond is stored in unpredictable order

AdamElliott-AB, 2021-11-16:
Sequence of events:
1. Packets are received by Yamcs in order with sub-millisecond generation time data
2. Yamcs records generation time in milliseconds (sub-millisecond precision is lost)
3. Yamcs stores parameter data, but the order of any data within 1 millisecond may not match the order received

#3 is a change from previous behavior. Parameter data within 1 millisecond used to be stored in the order received

Tested with master branch from 11/06/2021 233b5f77b4e2baeaa43844a44579971a0b328141

xpromache, 2021-11-16:
"previous behaviour" = which version?

AdamElliott-AB, 2021-11-16:
5.5.2

xpromache, 2021-11-16:
if you retrieve data with source=reply is the order correct?

https://docs.yamcs.org/yamcs-http-api/parameter-archive/get-parameter-samples/

AdamElliott-AB, 2021-11-16:
Yes. The order appears to be correct when using source=replay. 

I will need to do more testing, but I think this solution will be sufficient for the affected data.

xpromache, 2021-11-16:
Hmm, that solution always performs a replay (reprocessing the data) so not
very performant.

I think it only sorts correctly based on the CCSDS sequence count. But that
one wraps around so it might not always be correct (i.e. if the sequence
count wraps around in the same millisecond). I think previously it wasn't
correct either in these (probably very rare) cases.

I will try to see if I can make it reproduce the previous behaviour.

AdamElliott-AB, 2021-11-16:
Hmm, I am seeing this same out-of-order issue when monitoring a Parameter subscription as well. That seems strange.

I am pretty sure this is a change in behavior because I run a set of API tests each time we upgrade and am getting failures this time around.

AdamElliott-AB, 2021-11-16:
@xpromache I have just tracked this issue to a local change that I made. This is NOT an issue with Yamcs. I am deeply sorry for taking your time on this.

I was simulating communication delay (RF + ground segment) for the downlink. Packets within the same ms were incorrectly being released by my delay simulator to Yamcs in unpredictable order.

closed 2021-11-16 by AdamElliott-AB