yamcs/yamcs #636: listParameterHistory returns no data for aggregate parameter if start is < 1st data timestamp

AdamElliott-AB, 2021-11-22:
I think this line should call ParameterArchive.getIntervalEnd(
https://github.com/yamcs/yamcs/blob/4082ce0c561bec126458b1121213d8194e7d9b35/yamcs-core/src/main/java/org/yamcs/parameterarchive/MultiSegmentIterator.java#L64

leighgarbs, 2022-01-30:
You're referring to [this HTTP API call](https://docs.yamcs.org/yamcs-http-api/parameter-archive/list-parameter-history/)?

xpromache, 2022-01-31:
> I think this line should call ParameterArchive.getIntervalEnd(
> 
> https://github.com/yamcs/yamcs/blob/4082ce0c561bec126458b1121213d8194e7d9b35/yamcs-core/src/main/java/org/yamcs/parameterarchive/MultiSegmentIterator.java#L64

Sorry, I haven't noticed this issue until now when Leigh made a comment  about it :(

Your observation that we should change that to getIntervalEnd(req.stop) appears correct.

However, I do not see how that causes the problem in the subject which seems to be rather linked to the rangeStart than the rangeStop.

AdamElliott-AB, 2022-01-31:
Yes, this issue was found with the ListParameterHistory API call. We saw it when viewing the "Historical Data" tab of the parameter web view with a custom date range on an aggregate parameter. The issue did not appear with non-aggregate parameters. @xpromache , the req.stop value likely plays a part as well, but the error case we saw varied the req.start while keeping req.stop the same:
- req.start before start of data and req.stop after start of data matched no segment and returned no data
- req.start after start of data and req.stop after start of data matched a segment and did return data as expected
Changing the indicated line to use getIntervalEnd corrected the issue for us.

xpromache, 2022-01-31:
In fact I think we need to put just req.stop (without getIntervalEnd()). Please have a look at the diagram below.

The data is inserted in the archive with the segment start always as key with all data from a segment stored inside the same database record.

In order to retrieve data between req.start and req.stop, we need to get segments 2 and 3. However because we do not know where the segment 2 starts, we practically receive from the beginning of the interval (knowing that a segment cannot span across intervals!). In this case we will also retrieve segment 1 and it will be filtered out.

However, for the stop we need not go to the end of the interval because we know that any segment whose start is greater than req.stop, will contain no data from our request. 

Therefore I think it is enough to use req.stop there. In the example from the diagram, it will avoid going over segment 4.

Is it easy for you to confirm with a test case?

thx
nicolae

[image]

AdamElliott-AB, 2022-01-31:
That solution makes sense to me @xpromache . I apologize, but it is not easy for me to write a test case for this at this time.

xpromache, 2022-12-12:
closing this as a duplicate of #685 which has been fixed

closed 2022-12-12 by xpromache