yamcs/yamcs #638: Any plans of adding <scope>test<scope> to yamcs-core project?

lorenzo-gomez-windhover, 2021-12-17:
Hello

Hope you are all doing well.

To the best of my understanding, adding `<scope>test<scope>` essentially exposes the classes defined in `src/test/java`. So at that point any maven projects that would like to use a class such as `AbstractIntegrationTest` can. I suspect that this is not exposed by design, but I was just curious.

xpromache, 2021-12-20:
Hello,

The right way to do that would be to extract the AbstractIntegrationTest
into a new maven subbmodule "yamcs-test". That could be imported with a
test scope whenever necessary.

Otherwise, if we do as you suggest, I think we would run into trouble with
the releases which do not package the test classes.

As for the plans, no, we weren't planning on doing since we were not aware
of that being useful to anyone.
 You can live the issue open, maybe one day when we have more time we will
do it.

lorenzo-gomez-windhover, 2021-12-20:
I understand :+1:. The reason I brought this up was because `AbstractIntegrationTest` could be useful for devs that are using `yamcs-client`  APIs and would like to run an instance of YAMCS when running their unit tests. Which is my case at the moment. But I can always create a unit-test instance of my own that suits my needs so it's not so bad.

Thanks for the response!

xpromache, 2022-01-18:
I created a new module "tests" which you can import into your project to have the AbstractIntegrationTest available.

closed 2022-01-18 by xpromache