yamcs/yamcs #639: yamcs-web configuration documentation does not include displayPath nor stackPath

merose, 2021-12-23:
The WebPlugin source code responds to the configuration options `displayPath` and `stackPath`. However, these are not documented in the web interface configuration page: https://docs.yamcs.org/yamcs-server-manual/web-interface/configuration/

Also, there appears to be two different ways of externalizing displays and stacks to the file system. Configuring a path for a bucket named `displays` or `stacks` in the `buckets` configuration section seems to do the same thing ad the `displayPath` and `stackpath` settings. Is there a preferred configuration method?

fqqb, 2021-12-23:
Both these options are deprecated, which is why they get automatically suppressed from the online docs. You can find the old help texts here: https://github.com/yamcs/yamcs/blob/master/yamcs-web/src/main/resources/org/yamcs/web/WebPlugin.yaml#L16

I would say configuring buckets in yamcs.yaml is preferable. I had in mind (at some point in the future) to make it possible (at instance level) to choose the applicable buckets for either displays or stacks (currently they must be called "displays" and "stacks"), and the displayPath/stackPath options do not go together with that very well.

merose, 2021-12-23:
Yes, I see that, thanks, sorry for the noise. I'll close this issue.

closed 2021-12-23 by merose