yamcs/yamcs #658: Does Yamcs SLE service support CLTU-THROW-EVENT?

merose, 2022-03-21:
I will go through the code when I have time, but if it's a quick answer, let me know if Yamcs and yamcs-sle support the CLTU-THROW-EVENT operation for the SLE Forward CLTU service. The DSN now recommends missions use CLTU-THROW-EVENT to modify some parameters such as a data rate change, in order not to have to involve a DSN operator.

As an example, the CCSDS Forward CLTU Service recommendation (https://public.ccsds.org/Pubs/912x1b4.pdf) gives as an example, "For example, event-identifier may be ‘bit-rate-change’. Then event-qualifier may specify the new bit rate." (Section 3.11.2.7)

See also the examples of `event-identifier` and `event-qualifier` in Annex F of the same document. It looks like both of those parameters are envisioned to be strings values. It would be sufficient to support CLTU-THROW-EVENT from scripts only, at least at first.

xpromache, 2022-03-21:
I see it's in the jsle but not in yamcs-sle. The signature is this:

 throwEvent(int eventIdentifier, byte[] eventQualifier)

The definition of the eventIdentifier and eventQualifier is not part of the CCSDS standard but the standard lists some suggestions in annex F page 151.
https://public.ccsds.org/Pubs/912x1b4.pdf

I guess we could make it available via REST.

[image]

xpromache, 2022-03-25:
Implemented the GET-PARAMETER and CLTU-THROW-EVENT in the jsle and yamcs-sle.

One thing that could be improved is in the UdpFrameSink (part of the UDP to SLE bridge) to actually change the bitrate  (which  is a simple bits/sec) but I'm not sure how to interpret the parameters to the throw event in conjunction with the CltuParameters (subcarrierToBitRateRatio, modulationFrequency, modulationIndex).

closed 2022-03-25 by xpromache

xpromache, 2022-10-11:
I see it's in the jsle but not in yamcs-sle. The signature is this:

 throwEvent(int eventIdentifier, byte[] eventQualifier)

The definition of the eventIdentifier and eventQualifier is not part of the
CCSDS standard but the standard lists some suggestions in annex F page 151.
https://public.ccsds.org/Pubs/912x1b4.pdf

I guess we could make it available via REST.

[image: image.png]