yamcs/yamcs #667: Data links should all support setting the parent link to facilitate aggregation

merose, 2022-04-24:
For some ground station providers it is useful to group links under a common parent, using a custom aggregated data link class. However, not all link implementation classes support setting the parent link.

These classes apparently do not implement a way to set the parent link:
- TcpTmDataLink
- UdpTmDataLink
- FilePollingTmDataLink
-  UdpTmFrameLink
- The SLE data links from yamcs-sle

(The do get the default implementation in Link. However, the default methods in that interface ignore setting the parent link.)

Further, it would be useful to have an aggregated, or "grouped" data link class available to facilitate grouping of data links into a hierarchy. This is useful when there are multiple ground stations, each with several links (as when talking through the DSN, e.g.).

merose, 2022-04-24:
It might make sense to implement this in AbstractLink, since all data links seem to derive from that concrete class. Maybe also remove the default implementations of getParent() and setParent() in the LInk interface, leaving the definitions in the interface, but with no implementation.

merose, 2022-04-24:
Also, I've found that if you implement aggregated data links, it might be important to have another value in Status, for something like "partial" availability, which might show up as a yellow ball in the web UI. That is, if you have an aggregated link that connects both telemetry and telecommand, but only one of the child links is available, then the parent might show partial availability. (Right now I'm just showing the parent as Status.UNAVAIL in that case.)

merose, 2022-04-24:
As further discussion, we have found two new types of aggregated links to be useful:

1. A "group" data link, where enabling the parent enables all children in parallel.
2. A "round-robin" data link, where enabling the parent starts a process where each child is enabled in order. If a child does not connect within a retry interval, it is disabled and the parent moves on to try the next child. (This is important for ground systems like the DSN, where each ground station has several possible endpoints. Clients are supposed to try each endpoint in turn until one connects.)

xpromache, 2022-07-27:
Also belonging to refactoring the Links discussion: one annoying thing currently is that the command post-processor runs inside the Link.

If you have two ground stations with two links, you will have two post-processors which run independently. That will mess up the CCSDS sequence count generation.