yamcs/yamcs #672: Two algorithms cannot write the same parameter

merose, 2022-05-10:
If you configure two different algorithms to write the same result parameter, one algorithm is never fully initialized and will never be called.

## To reproduce ##

1. Add the SpaceSystem below at the end of src/main/yamcs/mdb/xtce.xml in the quickstart project.
2. Start the quickstart project using "mvn yamcs:run".
3. Browse to the parameter /myproject/DoubleAlgorithm/x.
4. Press "Set Value" in the UI and enter a value for "x", say 12. Expected and actual result: x is set to 12 and the parameter "result" is set to 24.
5. Navigate to the parameter /myproject/DoubleAlgorithm/y, and use "Set Value" to change its value to 23. Expected result: "y" changes to 23 and "result" changes to 46. Actual result: "result" variable is unchanged.

## Discussion ##

The XTCE added to xtce.xml defines three variables, "result", "x", and "y", and two MathAlgorithms, both of which indicate "result" as the result variable. One Yamcs starts, the first algorithm "SetFromX" displays normally in the Algorithms screens, but the second algorithm "SetFromY" is missing some data:
[image]

This could indicate that the algorithm was not initialized correctly, not sure.

### Why we need this functionality ###

In different modes our spacecraft sends the same data in different packet types, sometimes compressed in a different way, for example, to save bandwidth. We consolidate some of these parameters into new aggregate parameters. So it is inconvenient if two different algorithms cannot write the same local, output parameter.

## XTCE ##

This is the XTCE to add to the bottom of src/main/yamcs/mdb/xtce.xml, just before the closing "</SpaceSystem>".

~~~
	<SpaceSystem name="DoubleAlgorithm">
		<TelemetryMetaData>
			<ParameterTypeSet>
				<FloatParameterType name="Real"/>
			</ParameterTypeSet>
			<ParameterSet>
				<Parameter parameterTypeRef="Real" name="result"/>
				<Parameter parameterTypeRef="Real" name="x">
					<ParameterProperties dataSource="local"/>
				</Parameter>
				<Parameter parameterTypeRef="Real" name="y">
					<ParameterProperties dataSource="local"/>
				</Parameter>
			</ParameterSet>
			<AlgorithmSet>
				<MathAlgorithm name="SetFromX">
					<MathOperation outputParameterRef="result">
						<ParameterInstanceRefOperand parameterRef="x"/>
						<ValueOperand>2</ValueOperand>
						<Operator>*</Operator>
						<TriggerSet>
							<OnParameterUpdateTrigger parameterRef="x"/>
						</TriggerSet>
					</MathOperation>
				</MathAlgorithm>
				<MathAlgorithm name="SetFromY">
					<MathOperation outputParameterRef="result">
						<ParameterInstanceRefOperand parameterRef="y"/>
						<ValueOperand>2</ValueOperand>
						<Operator>*</Operator>
						<TriggerSet>
							<OnParameterUpdateTrigger parameterRef="y"/>
						</TriggerSet>
					</MathOperation>
				</MathAlgorithm>
			</AlgorithmSet>
		</TelemetryMetaData>
	</SpaceSystem>
~~~

merose, 2022-05-11:
We've worked around this by changing the parameter encoding to be uniform. But it is still something to address at some point. At the least, if this is a Yamcs restriction by design, some warning message would be appropriate in the log. As is is, it's somewhat mysterious behavior since there are no indications in the log that one algorithm will be ignored.

fqqb, 2023-03-17:
This was fixed on master, and in latest release (5.7.10).
This limitation was once added out of caution, but I see nothing wrong with the provided use case.

Have also made some UI updates to allow seeing the definition of a MathAlgorithm.

closed 2023-03-17 by fqqb