yamcs/yamcs #680: List Parameter History API with time range does not return parameter values at the "start" endpoint

merose, 2022-05-28:
The List Parameter History API appears to return only parameter values *after* the specified start time.

To reproduce:

1. In the quickstart project, add the SpaceSystem below to src/main/yamcs/mdb/xtce.xml, just before the closing "</SpaceSystem>".
2. Start Yamcs via "mvn yamcs:run".
3. In the Yamcs web UI, set a value for the local parameter /myproject/Test/sample.
4. Set another value for /myprject/Test/sample.
5. Using the List Parameter History API, get the two values: `$ curl 'http://localhost:8090/api/archive/myproject/parameters/myproject/Test/sample'`
6. Use the curl command again, but specify a "start" date equal to the first value time, e.g.: `$ curl 'http://localhost:8090/api/archive/myproject/parameters/myproject/Test/sample?start=2022-05-28T16:29:21.403Z'`

Expected result: both parameter values are returned

Actual result: only the second parameter value is returned

Addition to src/main/yamcs/mdb/xtce.xml:
~~~
	<SpaceSystem name="Test">
		<TelemetryMetaData>
			<ParameterTypeSet>
				<FloatParameterType name="SampleType" sizeInBits="64"/>
			</ParameterTypeSet>
			<ParameterSet>
				<Parameter parameterTypeRef="SampleType" name="sample">
					<ParameterProperties dataSource="local"/>
				</Parameter>
			</ParameterSet>
		</TelemetryMetaData>
	</SpaceSystem>
~~~

Examples of using the two curl commands in steps 5 and 6:

~~~
$ curl 'http://localhost:8090/api/archive/myproject/parameters/myproject/Test/sample'
{
  "parameter": [{
    "id": {
      "name": "sample",
      "namespace": "/myproject/Test"
    },
    "engValue": {
      "type": "DOUBLE",
      "doubleValue": 2.0
    },
    "generationTime": "2022-05-28T16:30:32.852Z",
    "acquisitionStatus": "ACQUIRED"
  }, {
    "id": {
      "name": "sample",
      "namespace": "/myproject/Test"
    },
    "engValue": {
      "type": "DOUBLE",
      "doubleValue": 1.0
    },
    "generationTime": "2022-05-28T16:29:21.403Z",
    "acquisitionStatus": "ACQUIRED"
  }]
}$ curl 'http://localhost:8090/api/archive/myproject/parameters/myproject/Test/sample?start=2022-05-28T16:29:21.403Z'
{
  "parameter": [{
    "id": {
      "name": "sample",
      "namespace": "/myproject/Test"
    },
    "engValue": {
      "type": "DOUBLE",
      "doubleValue": 2.0
    },
    "generationTime": "2022-05-28T16:30:32.852Z",
    "acquisitionStatus": "ACQUIRED"
  }]
}$
~~~

merose, 2022-05-28:
Note that if "order=asc" is specified, the start time is decremented by one in ParameterArchiveApi.java, and the query works correctly. (Unless this is the designed behavior for descending queries?)

merose, 2022-05-28:
It seems like these lines 427-429 in ParameterArchiveApi.java should have `if (pv.getGenerationTime() < start) {break;}`.

~~~
                if (pv.getGenerationTime() <= start) {
                    break;
                }
~~~

xpromache, 2022-05-30:
The queries from archive are supposed to return the interval [start, stop) and [stop, start) for ascending respectively descending queries. That is so you can do then another query starting at stop (ascending) or at start (descending) and not get duplicates.

merose, 2022-05-30:
I understand. I will talk with the OpenMCT developers to see how they want to work around this behavior. (The problem in their UI occurs when showing data in a table. If the user requests to change the time sorting order, they repeat the archive query, but this time they do not get the exact same result because of the endpoint behavior. Knowing exactly how it is designed to work may allow them a reasonable workaround.)

Will close this issue since the API is working as designed.

closed 2022-05-30 by merose