yamcs/yamcs #683: TcPacketHandler does not allow command postprocessor to change the length of the packet

merose, 2022-05-29:
In most classes that call a command postprocessor to generate the final, binary form of the command, the caller does not assume that the new, binary form is the same length as the original. However, TcPacketHandler reuses the original data array:

TcPacketHandler, lines 105-116.

~~~
        byte[] data = tf.getData();
        int offset = tf.getDataStart();
        for (PreparedCommand pc1 : l) {
            byte[] binary = cmdPostProcessor.process(pc1);
            if (binary == null) {
                log.warn("command postprocessor did not process the command");
                continue;
            }
            int length = binary.length;
            System.arraycopy(binary, 0, data, offset, length);
            offset += length;
        }
~~~

If the command postprocessor returns a longer array, the `System.arraycopy()` on the third to last line generates an `ArrayIndexOutOfBoundsException`.

xpromache, 2022-05-30:
The frame length is pre-computed and cannot be exceeded. This is because we have to make sure to not exceed the maximum size for one frame so we pick commands from the queue as long as the maximum is not exceeded.

The postprocessor getBinaryLength() method is used to get the binary length before the post-processor is run (this is a limitation but most of the time the post-processor will add some fixed number of bytes to the command and it can return the resulting length without actually doing the post-processing):
https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/main/java/org/yamcs/tctm/ccsds/TcPacketHandler.java#L82

So my guess is that in your case the getBinaryLength() returns the original length different than the length of the binary after post-processing.

merose, 2022-05-30:
Ah, I see. I did not realize I needed to override that method. Thank you. Will close this issue.

closed 2022-05-30 by merose