yamcs/yamcs #686: Yamcs client ParameterSubscription class has no way to set instance and processor

merose, 2022-06-02:
The ParameterSubscription class `add()` method allows adding parameter identifiers to a subscription, but leaves the instance and processor unset, resulting in an error return from the server over the websocket. The method source is this:

ParameterSubscription.java lines 64-75:
~~~
    /**
     * Extends the ongoing subscription with the provided identifiers.
     */
    public void add(List<NamedObjectId> ids) {
        if (!ids.isEmpty()) {
            clientObserver.next(SubscribeParametersRequest.newBuilder()
                    .setAction(Action.ADD)
                    .setAbortOnInvalid(false)
                    .addAllId(ids)
                    .build());
        }
    }
~~~

I was able to successfully subscribe by creating a SubscribeParametersRequest manually and sending using the subscription:
~~~
        SubscribeParametersRequest req = SubscribeParametersRequest.newBuilder()
                .setInstance(instance)
                .setProcessor(processor)
                .setAction(Action.ADD)
                .setAbortOnInvalid(true)
                .addAllId(ids)
                .build();
        subscription.sendMessage(req);
~~~

I think there should be a way to add the instance and processor, or perhaps the create subscription method should reside in the ProcessorClient instead.

fqqb, 2022-06-02:
Your second code is indeed how it is currently intended to work: a first message must be sent following the initial connection, which will activate the subscription. This is done as a separate step so that there is an opportunity for clients to attach message listeners before actually activating the subscription.

The first message must specify instance/processor, the action field is ignored for the first message.

If the client sends further messages (add/remove), the instance/processor remains the same as the initial message.

I agree there may be some improvement to be had here in terms of usability, just saying how it currently works. See also sample usage: https://github.com/yamcs/yamcs/blob/master/tests/src/test/java/org/yamcs/tests/ParameterTest.java#L85

merose, 2022-06-02:
Thanks, Fabian, I understand now. Will close this issue since it is working as expected.

closed 2022-06-02 by merose