yamcs/yamcs #697: List Parameter History API returns a value outside the search range in some conditions

merose, 2022-07-03:
To reproduce:

1. Clone the Yamcs quickstart repository.
2. Update the pom.xml to Yamcs version 5.7.4.
3. Add the test SpaceSystem below to the end of src/main/yamcs/mdb/xtce.xml, just before the closing `</SpaceSystem>`.
4. Start Yamcs via `mvn yamcs:run`.
5. Run the attached Python program `add_values.py` via `$ python3 add_values.py`. (Note: GItHub does not like attaching .py files, so it is attached as .py.txt.) The Python program adds 10 values for the local parameter `value`, one at 2023-11-25T15:00:01, and 9 starting at time 2023-11-25T15:00:52 and add 1 second intervals thereafter.
6. Run the curl command below.

~~~
$ curl 'http://localhost:8090/api/archive/myproject/parameters/myproject/Test/value?start=2023-11-25T15:00:00Z&stop=2023-11-25T15:00:02Z&order=asc'
~~~

Expected result: One value is returned with timestamp 15:00:01.

Actual result: Two values are returned, the value at 15:00:01 and another at 15:00:53. (It is unclear why that value is returned, since it is not the next value chronologically.)

(Notes: Using `order=desc` instead does not show the problem. If you restart Yamcs and run the curl command again, the problem goes away. It is only present so long as Yamcs is still running following adding the values, so it may be a problem with query processing using the RealtimeArchiveFiller.)

The values in Yamcs can be seen in this view of historical data from the Yamcs web UI:

[image]

## The additional SpaceSystem

Add this SpaceSystem to src/main/yamcs/mdb/xtce.xml just before the closing `</SpaceSystem>`.

~~~
  <SpaceSystem name="Test">
    <TelemetryMetaData>
      <ParameterTypeSet>
        <IntegerParameterType name="Value" sizeInBits="32">
          <IntegerDataEncoding sizeInBits="32" byteOrder="mostSignificantByteFirst" encoding="twosComplement"/>
        </IntegerParameterType>
      </ParameterTypeSet>
      <ParameterSet>
        <Parameter parameterTypeRef="Value" name="value">
          <ParameterProperties dataSource="local"/>
        </Parameter>
      </ParameterSet>
    </TelemetryMetaData>
  </SpaceSystem>
~~~

## The Python program

The program `add_values.py` is attached as `add_values.py.txt` because of GitHub restrictions. Run the program via:

    $ python3 add_values.py.txt

[add_values.py.txt](https://github.com/yamcs/yamcs/files/9035596/add_values.py.txt)

## Discussion

- It appears that the timestamps need to cross a segment boundary for the problem to occur.
- The number of values in the following segment needs to be at least 9. If you modify the Python program to store only 8 values, the problem does not occur. I do not know whether the number of values is related to the segment chunk size or not.
- The problem is only present while Yamcs is running. If Yamcs is restarted, the problem goes away.
- Specifying `order=desc` in the query does not exhibit the problem. It only occurs for ascending order queries.

merose, 2022-07-03:
This appears to be the same problem as https://github.com/yamcs/yamcs/issues/690, but this test case is smaller. I'll leave this open for now, but may consolidate the issues and close this one when I get a chance.

xpromache, 2022-12-12:
closing this as a duplicate of #690 which has been fixed.

closed 2022-12-12 by xpromache