yamcs/yamcs #702: Alarms on aggregate telemetry items

PilotWave, 2022-07-25:
Alarms don't seem to work on aggregate parameters.   I can get alarms to work on regular, non aggregate, parameters.

I have an aggregate parameter called System_Telemetry which contains:
{
temperature test0_temp_C;
temperature test1_temp_C;
temperature test2_temp_C;
temperature test3_temp_C;
temperature test4_temp_C;
temperature test5_temp_C;
}

I defined a basic low value alarm (<1, onValueChange) on System_Telemetry.test0_temp_C

the type temperature is a defined type which executes a java-expression to convert from byte to degC (float)
Confirmed these values are being updated and are out of range of the defined alarm

Running yamcs 5.7.5

xpromache, 2022-07-26:
In Yamcs alarms are at parameter level, we do not support alarms on aggregate members.

I guess something could be implemented by making some convention like the alarm state of the parameter is the maximum  alarm state of all members (i.e as soon as one member is out of limits an alarm is raised for the whole parameter).

PilotWave, 2022-07-26:
Thanks, that is what I was guessing.  That is a rather strange behavior though.  Everything else seems to be applied correctly on aggregate (struct) members (e.g. calibration).   I would have expected items of a stucture to be treated just like parameters.  For the most part, they seem to be.

We have a repeating set of telemetry items for multiple identical subsystems and so Inlining all the items would be pretty ugly since we would have to create multiple copies and name them all differently.

For now, the only workaround I found is to add alarms to derived parameters from each struct item, but that is not very nice.

xpromache, 2022-07-26:
Indeed, I admit that it looks like a strange behavior.

In Yamcs a Parameter Value is a quadruplet (time, raw value, engineering value, status). The status is the thing encoding out of limits information. 

The engineering value  and raw value for aggregate parameters are aggregate values  - the members of those aggregate values are things like integer or float - they are not parameter values themselves - so they do not have a status.

The way this could be made to work without too much of a big headache is to come up with an aggregate status (similar to the aggregate value) - a status that is split in components (which are themselves statuses). 

As for your concrete problem: 
It is possible to reuse container definitions if you use XTCE files.  Basically you have one file that defines the common structure, and then you refer to it multiple times in the mdb yaml definition. So you end up with 
/subsystem1/param1
/subsystem1/param2
/subsystem2/param1
/subsystem2/param2
...
where param1 and param2 are defined in the xtce file which is loaded multiple times.

From Yamcs point of view they will all be different parameters but you only defined it once.

You are definitely not the first one to have this problem so maybe we should find a better solution for it.

xpromache, 2022-07-26:
Actually if you use xls files it's also possible the same way I think. You just put all your common parameters into one xls file and you refer to it multiple times in the mdb yaml definition.

PilotWave, 2022-07-26:
Thanks, that is a good suggestion.  There was a reason why we didn't go the subsystem way but that escapes me now.  I will revisit this option.