yamcs/yamcs #704: Continuous telemetry on SLE connection causes OutOfMemoryError on "Direct buffer memory" in Yamcs

merose, 2022-07-26:
I've found an issue that I have not fully characterized. It appears that there is a memory leak within the Netty layer or the SLE layer in Yamcs. I believe it does not occur when using a UDP frames connection, only an SLE connection, but I will verify this. Currently we are using a Return All Frames (RAF) connection for telemetry reception. When receiving telemetry continuously at an average rate of about 400k bit/sec, we see the error below after about 26 hours of operation.

This is not an out of heap situation, apparently. Make be mitigated by adding directory buffer memory to Java. (Default is only 64Mb.) Or may indicate a resource leak.

I will add more info as I characterize it further.

~~~
Jul 08 17:28:26.557 _global [285] org.yamcs.sle.user.AbstractServiceUserHandler 
[WARNING] Caught exception Direct buffer memory
Jul 08 17:28:26.559 viper [285] org.yamcs.sle.TmSleLink[localhost1-RAF] [WARNING] SLE exception caught: java.lang.OutOfMemoryError: Direct buffer memory
        java.base/java.nio.Bits.reserveMemory(Bits.java:175)
        java.base/java.nio.DirectByteBuffer.<init>(DirectByteBuffer.java:118)
        java.base/java.nio.ByteBuffer.allocateDirect(ByteBuffer.java:317)
        io.netty.buffer.PoolArena$DirectArena.allocateDirect(PoolArena.java:648)
        io.netty.buffer.PoolArena$DirectArena.newChunk(PoolArena.java:623)
        io.netty.buffer.PoolArena.allocateNormal(PoolArena.java:202)
        io.netty.buffer.PoolArena.tcacheAllocateSmall(PoolArena.java:172)
        io.netty.buffer.PoolArena.allocate(PoolArena.java:134)
        io.netty.buffer.PoolArena.allocate(PoolArena.java:126)
        io.netty.buffer.PooledByteBufAllocator.newDirectBuffer(PooledByteBufAllocator.java:395)
        io.netty.buffer.AbstractByteBufAllocator.directBuffer(AbstractByteBufAllocator.java:188)
        io.netty.buffer.AbstractByteBufAllocator.directBuffer(AbstractByteBufAllocator.java:179)
        io.netty.buffer.AbstractByteBufAllocator.ioBuffer(AbstractByteBufAllocator.java:140)
        io.netty.channel.DefaultMaxMessagesRecvByteBufAllocator$MaxMessageHandle.allocate(DefaultMaxMessagesRecvByteBufAllocator.java:120)
        io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:150)
        io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:722)
        io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:658)
        io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:584)
        io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
        io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)
        io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
        io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
        java.base/java.lang.Thread.run(Thread.java:829)Jul 08 17:28:26.560 _global [285] org.yamcs.events.EventProducer [FINE] event: INFO; AbstractTmSleLink$MyMonitor; SLE exception caught: Direct buffer memory 
~~~

merose, 2022-07-26:
Once the OutOfMemory error occurs, the Yamcs UI is nonresponsive, though the process remains running.

xpromache, 2022-07-26:
Can you start Yamcs with --netty-leak-detection option? 

Maybe we will get a hint where the leak is.

merose, 2022-07-26:
Yes, will do and let you know. Can also supply sample telemetry and XTCE if needed, but will start by capturing additional info with that flag.

merose, 2022-07-27:
There are a lot of ResourceLeakDetector log messages that look are similar to the one below. Over 30,000 such log messages within about 10 minutes. I can provide log files if it is helpful.

~~~
Jul 26 22:35:13.464 _global [87] io.netty.util.ResourceLeakDetector [SEVERE] LEAK: ByteBuf.release() was not called before it's garbage-collected. See https://netty.io/wiki/reference-counted-objects.html for more information.
Recent access records: 
#1:
	io.netty.buffer.AdvancedLeakAwareByteBuf.readBytes(AdvancedLeakAwareByteBuf.java:503)
	io.netty.buffer.ByteBufInputStream.read(ByteBufInputStream.java:183)
	com.beanit.jasn1.ber.internal.Util.readFully(Util.java:28)
	com.beanit.jasn1.ber.internal.Util.readFully(Util.java:23)
	com.beanit.jasn1.ber.types.BerOctetString.decode(BerOctetString.java:76)
	ccsds.sle.transfer.service.raf.outgoing.pdus.RafTransferDataInvocation.decode(RafTransferDataInvocation.java:342)
	ccsds.sle.transfer.service.raf.outgoing.pdus.FrameOrNotification.decode(FrameOrNotification.java:121)
	ccsds.sle.transfer.service.raf.outgoing.pdus.RafTransferBuffer.decode(RafTransferBuffer.java:115)
	org.yamcs.sle.user.RafServiceUserHandler.processData(RafServiceUserHandler.java:137)
	org.yamcs.sle.user.AbstractServiceUserHandler.channelRead(AbstractServiceUserHandler.java:138)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	org.yamcs.sle.Isp1Handler.channelRead(Isp1Handler.java:60)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:324)
	io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:311)
	io.netty.handler.codec.ByteToMessageDecoder.callDecode(ByteToMessageDecoder.java:432)
	io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:276)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:722)
	io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:658)
	io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:584)
	io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
	io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)
	io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	java.base/java.lang.Thread.run(Thread.java:829)
#2:
	io.netty.buffer.AdvancedLeakAwareByteBuf.readByte(AdvancedLeakAwareByteBuf.java:401)
	io.netty.buffer.ByteBufInputStream.read(ByteBufInputStream.java:172)
	com.beanit.jasn1.ber.BerLength.decode(BerLength.java:106)
	com.beanit.jasn1.ber.types.BerOctetString.decode(BerOctetString.java:71)
	ccsds.sle.transfer.service.raf.structures.AntennaId.decode(AntennaId.java:111)
	ccsds.sle.transfer.service.raf.outgoing.pdus.RafTransferDataInvocation.decode(RafTransferDataInvocation.java:315)
	ccsds.sle.transfer.service.raf.outgoing.pdus.FrameOrNotification.decode(FrameOrNotification.java:121)
	ccsds.sle.transfer.service.raf.outgoing.pdus.RafTransferBuffer.decode(RafTransferBuffer.java:115)
	org.yamcs.sle.user.RafServiceUserHandler.processData(RafServiceUserHandler.java:137)
	org.yamcs.sle.user.AbstractServiceUserHandler.channelRead(AbstractServiceUserHandler.java:138)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	org.yamcs.sle.Isp1Handler.channelRead(Isp1Handler.java:60)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:324)
[... cut ...]
	io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:496)
	io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:986)
	io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	java.base/java.lang.Thread.run(Thread.java:829)
: 22 leak records were discarded because the leak record count is targeted to 4. Use system property io.netty.leakDetection.targetRecords to increase the limit.
~~~

xpromache, 2022-07-27:
I guess that's why the settings is called PARANOID :)

Will investigate and fix shortly, thanks for the log.

xpromache, 2022-07-27:
I just pushed some changes to jsle, can you give it a try?

merose, 2022-07-27:
Trying today. Note that I found an unrelated minor problem in unit test in jsle: https://github.com/yamcs/jsle/issues/10

Doesn't fail for you since you're east of Greenwich. :-)

merose, 2022-07-27:
Preliminary results indicate no ResourceLeakDetector warnings/errors at all in the log. Will let it run for a while and confirm, and then will run a 24hr test.

merose, 2022-07-27:
After running for about 15 minutes, there are zero ResourceLeakDetector messages in the log, where there were over 30,000 previously. Will now test this overnight to verify that it works longer-term.

merose, 2022-07-28:
Has been running for several hours now with continuous telemetry. No ResourceLeakDetector messages. I believe this has fixed the problem. Will there be a release of jsle?

xpromache, 2022-07-28:
Yes, I will make a new release shortly.

closed 2022-08-05 by xpromache