yamcs/yamcs #705: Exception after upgrade to 5.7.5

PilotWave, 2022-07-27:
Seems to be dependency related but I cannot figure this out.  I am getting these exceptions when I send commands after switching to v5.7.5.  Looks like it doesn't see the ByteArrayUtils package but I don't know how to add it.

11:39:09.089 _global [19] YamcsServer Uncaught exception 'java.lang.NoSuchMethodError: 'byte[] org.yamcs.utils.ByteArrayUtils.encodeShort(int, byte[], int)'' in thread Thread[Thread-2,5,main]: [com.example.myproject.MyCommandPostprocessor.process(MyCommandPostprocessor.java:57), org.yamcs.tctm.AbstractTcDataLink.postprocess(AbstractTcDataLink.java:92), org.yamcs.tctm.UdpTcDataLink.uplinkCommand(UdpTcDataLink.java:52), org.yamcs.tctm.AbstractThreadedTcDataLink.run(AbstractThreadedTcDataLink.java:131), java.base/java.lang.Thread.run(Thread.java:829)]
11:39:09.090 _global [19] CrashHandler type: UncaughtException, msg: Uncaught exception 'java.lang.NoSuchMethodError: 'byte[] org.yamcs.utils.ByteArrayUtils.encodeShort(int, byte[], int)'' in thread Thread[Thread-2,5,main]: [com.example.myproject.MyCommandPostprocessor.process(MyCommandPostprocessor.java:57), org.yamcs.tctm.AbstractTcDataLink.postprocess(AbstractTcDataLink.java:92), org.yamcs.tctm.UdpTcDataLink.uplinkCommand(UdpTcDataLink.java:52), org.yamcs.tctm.AbstractThreadedTcDataLink.run(AbstractThreadedTcDataLink.java:131), java.base/java.lang.Thread.run(Thread.java:829)]

xpromache, 2022-07-27:
I apologize but I removed that function because it was not specified if it works with signed or unsigned numbers :(

There is a encodeUnsignedShort instead, you have to change your code.

PilotWave, 2022-07-27:
you mean the command structure?  I am not directly using this code, I just send a command from the web interface and I get this error.   Right now if I sent an empty CCSDS command it still fails.

merose, 2022-07-27:
You're using the Yamcs quickstart project that has a command postprocessor class called MyCommandPostprocessor (https://github.com/yamcs/quickstart/blob/master/src/main/java/com/example/myproject/MyCommandPostprocessor.java). The version of that class that you're using is out of date and does not work with Yamcs 5.7.5. A change was committed on Mar 2nd that fixes it:

~~~
commit 6a05547d30 (HEAD -> master, origin/master, origin/HEAD)
Author: Fabian Diet <fabian.diet@spaceapplications.com>
Date:   Wed Mar 2 20:17:50 2022 +0100

    Use Yamcs v5.6.0
~~~

Update your copy of quickstart and you should be able to use Yamcs 5.7.5. I was able to pull quickstart and then edit pom.xml to update Yamcs, and it worked fine.

PilotWave, 2022-07-27:
Thanks so much, that worked.

closed 2022-07-27 by PilotWave