yamcs/yamcs #708: Question: A good way to clean the database?

nmaas87, 2022-08-05:
Is there a good way to clean the RocksDB from unneeded data? E.g. we got certain Yamcs instances running at all times, but are only using it with a EM/FM during certain periods. Yamcs is still recording its own health data etc so the database is way bigger than it needs to be. Any quick command to delete all data between datetime xx and datetime yy?

Thanks a lot :)

xpromache, 2022-08-08:
Hi Nico,

Do you want to remove the parameters or the packets or both?

First you should make a backup.

Then you can login with the sql command line tool "yamcs dbshell" and execute:

delete from tm where gentime < '2022-08-01T00:00:00Z' and gentime > '2022-01-01T00:00:00Z'

and/or the same with pp instead of tm.

You can execute a select first to see what would get removed.

After that you have to clean the parmeter archive which stores the same data but in a different format. For that you can use this call:
https://docs.yamcs.org/yamcs-http-api/parameter-archive/rebuild-range/

(I just now read the text above which says that the rebuild does not remove data - it's wrong, I will correct it)
You can make a json file with the start and stop and execute the command with curl.

nmaas87, 2022-08-09:
Hi Nicolae,
I guess I want to remove pp, as I want to get rid of the statistics of Yamcs written during non-OPS / "standby" time (like harddrive space available, jmvMemoryUsed etc.). Your answer is perfect. Is the rebuild of the PP archive actually a "bad thing" to do? Meaning if one were to do it over the complete time range - would that change data / have sideeffects (besides the CPU cycles)?

Thanks :)

xpromache, 2022-08-09:
No, you can rebuild the parameter archive over whatever time interval.
The only thing to remember is that it will practically do replay and reprocess data according to the currently configured MDB.

nmaas87, 2022-08-10:
Perfect! Thank you very much! :)

closed 2022-08-10 by nmaas87