yamcs/yamcs #715: TSE UDP Device Support

nmaas87, 2022-09-11:
Dear all,
is there any way to include support for an UDP TSE device?
(Meaning connecting via raw UDP to an TSE TM/TC for SCPI parsing?)
Thank you :)

fqqb, 2022-09-13:
Currently there's only these two driver types: 
* https://github.com/yamcs/yamcs/blob/master/yamcs-tse/src/main/java/org/yamcs/tse/SerialPortDriver.java
* https://github.com/yamcs/yamcs/blob/master/yamcs-tse/src/main/java/org/yamcs/tse/TcpIpDriver.java

A UDP driver should be straightforward to add. If nobody else picks it up, I'll look at it in the not so far away future.

nmaas87, 2022-09-13:
Thanks @fqqb that would be an awesome addition. Especially if one could configure the UDP to be client or server mode.
(e.g. being able to configure to which IP and Port to send raw UDP packets with the SCPI message and on which port to receive the answers on Yamcs)

nmaas87, 2023-02-25:
Hey @fqqb - any updates on this? :)
Thanks a lot!

xpromache, 2023-02-28:
> Hey @fqqb - any updates on this? :) Thanks a lot!

Hi Nico,

Triggered by a project where we have to control an instrument via MIL-STD-1553, we want to implement a more general Yamcs EGSE frontend which can control devices via hardware cards and which will also take over the current TSE functionality. The configuration of the current TSE is not very user friendly and that's another reason to refactor it.

However, the Yamcs EGSE frontend may take a while to materialize...

In the meanwhile, if you really need the SCPI on top of UDP we can maybe implement something quickly - but please provide more details: which direction do you need it to go?
Will Yamcs sending UDP packets containing SCPI messages to a predefined host:port be enough for you?

nmaas87, 2023-02-28:
That sounds like some cool feature set coming, nice.
Yes, sending to a UDP host:port and also receiving data from there (as the answer packets) is completely fine

fqqb, 2023-03-29:
Pull request #771 includes this functionality + doc update.

nmaas87, 2023-03-29:
Awesome, thank you very much @fqqb / Fabian! :) Will check it out when its in the release :)

closed 2023-04-03 by nmaas87