yamcs/yamcs #728: Question about decoding packets in a non-XTCE-friendly format

merose, 2022-10-11:
We have some data packets coming from a partner that is not amenable to direct parsing from XTCE definitions. Instead, it's in a self-describing format of (id,value) pairs. I'd like to use Yamcs to store those values as new values of parameters. I have a couple of possibilities in mind:

1. Create a new data link that sends the packets to a new stream name. Configure a new service to decode the packets into parameter values and write to another stream. Configure a StreamParameterProvider to listen to that stream and write to the ParameterProcessor.
2. Define the encoded portion of the packet as a single, binary parameter. Write a new algorithm to decode the binary parameter and set output parameters to the appropriate values.
3. Create a custom packet preprocessor that takes the original packet and processes it, but also pushes onto a stack additional packets with a single (id,value) in it, plus any header info necessary to identify it, and process those generated packets from the stack in the run() loop as if they had arrived from outside.

Option 1 seems most elegant, but involves doing things I haven't done before.

Option 2 is the simplest, but involves configuring a very large number of output parameters for the algorithm, only some of which will be written at any one time. (I don't know whether this will affect grouping of parameters in the RocksDB archive segments.)

Option 3 is most brute-force, and would involve creating many, many XTCE containers that are not received from outside, but from the packet preprocessing. The parameter data would be archived three times, once as the original packet, once as the generated "pseudo-packets", and once as the parameter values. It has the downside that you can't just download all the packets and replay them through the data link, because some didn't come from the data link to begin with.

Any suggestions on the above options, or any new suggestions I have not thought of would be greatly appreciated.

merose, 2022-10-11:
The more I look at it, the more option 1 appears to be similar to a combination of CfsEventDecoder (subscribe to packet stream) and SystemParametersProducer (produce a stream of tuples where the columns are ParameterValue objects). I think I can handle doing that if you all think this is the right approach.

xpromache, 2022-10-12:
Hi Mark, please have a look at the UdpParameterDataLink - it does that kind
of thing.

One thing to pay attention to when calling the
parameterSink -> updateParameters(long gentime, String group, int seqNum,
Collection<ParameterValue> params);

(this method puts the parameters on the pp stream)

The (gentime, group, seqNum) is used as a primary key - so if you provide
the same parameters with that triplet unchanged, they will not be saved.

Also, important for the parameter archive buildup is that the parameters
coming together should be in the same group or more importantly, the
parameters which do not come together all or most of the time should not be
in the same group.
For example if you have some device for which you sample every second 5
temperatures and 3 power statuses, you can have those 8 parameters in the
same group. If you have a second device for which you sample other
parameters and the sampling is not synchronized with the first device, then
you should have them in a second group.

nicolae

merose, 2022-10-12:
Is there any naming convention for groups so I can choose group names that are guaranteed not to conflict with existing groups?

xpromache, 2022-10-12:
I think we only use "/yamcs/<server_id>" for the system parameters.

You can use whatever you want; usually something representing the origin or
type of data.  You can use a "/a/b/c/" notation but it is not required.
For example in the ISS ground segment where these so-called "Processed
Parameters" are used a lot, a group would be simply "EDR" (EDR = European
Drawer Rack).

merose, 2022-10-12:
A couple more questions, about parameter types that are enumerations or that need a calibrator.

- In some cases the decoded parameter will be an integer of an enumerated type. For such a parameter do I just set the raw value in the ParameterValue? Will parameterSink.updateParameters() ensure the engineering value is filled in? Or do I have to do something else to get the conversion to the string label?
- If there needs to be a calibrator for a numeric parameter, do I just set the raw value and let parameterSink.updateParameters() ensure the calibrator is called, or do I need to do something else to generate the engineering value?

xpromache, 2022-10-12:
> I'd like to archive the packets as well as the encoded parameters. In
> addition, there is a prefix to each packet that *is* amenable to decoding
> through XTCE. So it looks like, based on your suggestion, that a new data
> link could do both operations, as follows:
>
>    - The new data link should implement both TmPacketDataLink *and*
>    ParameterDataLink, so that both setTmSink() and setParameterSink() are
>    called during link initialization.
>    - The new data link should call tmSink.processPacket() to ensure that
>    the packet is archived and the packet prefix is decoded according to the
>    XTCE.
>    - In addition, the new data link should decode the non-XTCE-friendly
>    portion of the packet into parameters and call
>    parameterSink.updateParameters() with those decoded parameters, using an
>    appropriate group name, as you note above.
>
> Does this sound correct?

yes, it sounds correct. Pls let me know if you encounter any issues.

merose, 2022-11-04:
Thanks, worked fine. The only thing I had trouble with was knowing to set the `ppStream: pp_realtime` configuration parameter on the data link, but became apparent when the number of columns in the tuple was wrong because of the wrong destination class. Will close this issue, thanks again.

closed 2022-11-04 by merose