yamcs/yamcs #729: Yamcs web UI changes all radio buttons in command arguments whenever one is changed

merose, 2022-10-18:
Environment:
- Linux Ubuntu 20.04
- Yamcs 5.7.8
- Chrome browser 106.0.5249.103

If a command has multiple arguments that are `<BooleanArgumentType>`, which are represented in the UI as radio buttons, changing one radio button changes the radio buttons for all arguments, in unison.

To reproduce:
1. Get the Yamcs quickstart project.
2. Add the "Test" SpaceSystem below to the end of src/main/yamcs/mdb/xtce.xml, just before the closing `</SpaceSystem>`.
3. Start Yamcs: `$ mvn yamcs:run`
4. Browse to http://localhost:8090
5. Go to Commanding > Send a command, and select the command /myproject/Test/SendFlags.
6. Click the link "Show 2 arguments with defaults". The two arguments _alpha_ and _beta_ should both be _false_.
7. Click the _True_ radio button for argument _alpha_.

Expected result: Only _alpha_ changes to _True_, _beta_ remains _False_.

Actual result: Both _alpha_ and _beta_ change to _True_.

Workaround: Use `<EnumeratedArgumentType>` instead of `<BooleanArgumentType>` with the same encoding. The downside is that this results in a dropdown in the UI, which requires two clicks to select instead of one.

-----

Add this SpaceSystem to src/main/yamcs/mdb/xtce.xml:

~~~
	<SpaceSystem name="Test">
		<CommandMetaData>
			<ArgumentTypeSet>
				<BooleanArgumentType name="FlagType">
					<IntegerDataEncoding sizeInBits="1"/>
				</BooleanArgumentType>
			</ArgumentTypeSet>
			<MetaCommandSet>
				<MetaCommand name="SendFlags">
					<BaseMetaCommand metaCommandRef="MyProjectPacket">
						<ArgumentAssignmentList>
							<ArgumentAssignment argumentName="Packet_ID" argumentValue="100" />
						</ArgumentAssignmentList>
					</BaseMetaCommand>
					<ArgumentList>
						<Argument argumentTypeRef="FlagType" name="alpha" initialValue="false"/>
						<Argument argumentTypeRef="FlagType" name="beta" initialValue="false"/>
					</ArgumentList>
					<CommandContainer name="SendFlagsContainer">
						<EntryList>
							<ArgumentRefEntry argumentRef="alpha"/>
							<ArgumentRefEntry argumentRef="beta"/>
						</EntryList>
					</CommandContainer>
				</MetaCommand>
			</MetaCommandSet>
		</CommandMetaData>
	</SpaceSystem>
~~~

fqqb, 2022-10-19:
Can reproduce. Looking into it.

I'll note that in your example `initialValue` should be set to `False` instead of `false`, as that is the default value of the `zeroStringValue` attribute of the `<BooleanArgumentType />` when unspecified (inherited from `<ArgumentBooleanDataType />`). Yamcs appears to be silent on this type of definition error, which I'll fix too.

fqqb, 2022-10-19:
Initial value targeted for 5.7.x.
Case sensitivity check targeted for 5.8.x.

fqqb, 2022-10-26:
On master, case-sensitive check became a deprecation warning for the time being.

closed 2022-10-26 by fqqb