yamcs/yamcs #731: Is there a System privilege to give access to the admin screens?

merose, 2022-10-19:
I can't find a system privilege that controls access to the admin pages. What we'd like is:

superuser MINUS ability to send commands, and maybe MINUS a few other privileges

This is for a system engineer role in the mission operations center. That role needs to be able to debug any problem, but should not be able to send commands. We currently have to use superuser privileges, but that gives all system and object permissions, so we have to trust that the engineer role will not inadvertently send a command.

fqqb, 2022-10-20:
The webapp explicitly checks for superuser attribute before granting access to any of the admin pages. This was done for simplicity reasons at the time.

Indeed would be good to make more flexible.

fqqb, 2023-03-23:
pull request #731 introduces a new privilege: `web.AccessAdminArea` which grants access to the Admin Area.

The functionalities within Admin Area are separated in the following groups:

| System Privilege | Pages |
| - | - |
| ControlAccess      | Access control, Sessions       |
| ControlArchiving  | Database, RocksDB     |
| ControlServices | Services |
| ReadSystemInfo |  Processor types, Replication, HTTP traffic, API routes, Leap seconds, Threads, System |