yamcs/yamcs #760: Frontend hh:mm:ss bounds for XtimeParameterType

Stavrenas, 2023-02-04:
As you can see bellow, the user can enter values larger than:
- 24 for the hours segment
- 60 for the minutes segment
- 60 for the seconds segment
I can send the command without any problem. If this is the intended behaviour, feel free to close the issue.
[image]
The resulting unix date is simply the starting date plus `hours * 24 * 60 * 60  + minutes * 60 * 60  + seconds * 60`, so it's **not** *invalid* or *illegal*.

xpromache, 2023-02-06:
Just a small remark: 60 is sometimes valid for seconds (when leap seconds are enabled). I know we try to avoid usage of leap-seconds in the front-end (by performing smearing) and moreover leap seconds will disappear in the future, but still maybe relevant for the past.

fqqb, 2023-07-05:
Closing because I quite like the rollover.

closed 2023-07-05 by fqqb