yamcs/yamcs #772: Feature requrst: Add CommandArgs option to enforce generation of checksum at the end of a TC

nmaas87, 2023-04-04:
With regard to: https://groups.google.com/g/yamcs/c/57apL7NWMh0
As Google just deleted my answers for a third time, I am posting it here.

In the past I used the wrong CRC (KERMIT) so I had to run a custom plugin in Yamcs. Half a year ago, I corrected the Checksum to CRC16-CCITT and now wanted to use it with stock-Yamcs. I think it would be awesome if "enforceCRCchecksumToTheLastBytes" could be set as a CommandArgs option. This would make Yamcs very flexible and allow it to adapt better to existing payloads (because adapting existing payloads to Yamcs will probably not happen and the secondary CCSDS header is extremely vague, all exploration payloads I worked with by design demanded the TC to have a CRC...).  It could also be added to the org.yamcs.tctm.GenericCommandPostprocessor if this fits better. I just need some Postprocessor to set the APID, the Type to TC and compute+set the Checksum. The first 2 were working already if I saw correctly, only the checksum is missing :)

xpromache, 2023-04-04:
If you are referring to the IssCommandPostprocessor - that one is implementing ISS way of CCSDS packets which mandate the bit in the secondary header. In my opinion, adding an option that will render the packets invalid (or at least not according to the spec) would not be a good way to go.
The ISS secondary headers contain more than just that bit - for example they contain also the time in a specific format on a specific position. I find dubious that a payload implements the time according to that spec but not the checksum bit.

The GenericPacketPostporcessor will always add the checksum if configured. But that one doesn't do CCSDS headers.

Perhaps what you want is a more generic CcsdsCommandPostprocessor. Could be added but frankly it's not something I would  do in my free time. Contributions are always welcome...

nmaas87, 2023-04-04:
Thanks for clearing that up!
Yeah the payload is not ISS specific but "normal" CCSDS, hence I did not know anything about the ISS specific implementation of the CCSDS standard.

Looks like really a CcsdsCommandPostprocessor would be the best solution for the issue. Really sorry, I was behind the blackbox of the plugin a colleague wrote for the payload and I saw seqCnt, TM type, checksum just working and assumed only the "wrong" CRC checksum was done by the plugin (which was not CCSDS compatible as being KERMIT...) and Yamcs would be able to speak generic CCSDS directly. Sorry for the confusion, I only got a month left for a small conference talk about Yamcs and it is very good that this comes up now, so I am not saying anything wrong about the capabilities of Yamcs CCSDS encoding/decoding. I think for the time being I will use the plugin and change the CRC checksum to be the correct one to get the presentation working.

xpromache, 2023-04-04:
CCSDS is an organization which developed lots of standards.  "CCSDS encoding/decoding" in general does not mean much.
One of the standards developed by CCSDS is the [Space Packet Protocol](https://public.ccsds.org/Pubs/133x0b2e1.pdf) which defines the Space Packets colloquially called in our ISS world "CCSDS Packets".

The CCSDS Space Packet standard defines only the primary header including one bit "secondary header present". It does not define any particular secondary header - that is left for specific missions to define.
ESA has developed the PUS standard including a specific definition of the secondary header. cFS (a popular framework for implementing space software) has another definition. 

ISS predating all these, have their own definition which is not even exactly matching the CCSDS standard because the Packet Type bit in the primary header supposed to be used as 0=TC, 1=TM, is used by ISS packets for extending the APID.

So unless your conference talk is specifically targeted towards ISS world, you need to better define what "CCSDS encoding/decoding" means.