yamcs/yamcs #775: Alarms updated at runtime

shefalijoshi, 2023-04-18:
Open MCT shows limits as horizontal lines in our plots. To do this we get the limits returned by a query to /api/mdb for the relevant parameter.
For various reasons we also need to update limits at runtime, but we have discovered that subsequent queries to /api/mdb return the limits as persisted, and does not include any runtime overrides.

How would you recommend that we establish the runtime limits for a plot when it loads?

Also, is there some way of knowing when a runtime limit has changed? Like a change stream of some kind? Bear in mind that these plots may not always have telemetry flowing, depending on how data down-link is being prioritized in a given mission phase, so we cannot rely on the limits provided with parameter values via web sockets etc.

akhenry, 2023-05-15:
@xpromache I think this solution would work, thank you.

shefalijoshi, 2023-06-07:
@xpromache Let us know if there's anything else you need to proceed with this. Thank you.

fqqb, 2023-06-08:
Shall be in next release, which should be soon

fqqb, 2023-06-12:
API update available as of v5.8.2:
https://github.com/yamcs/yamcs/releases/tag/yamcs-5.8.2

* List: https://docs.yamcs.org/yamcs-http-api/mdb-override/list-mdb-overrides/
* Subscribe: https://docs.yamcs.org/yamcs-http-api/mdb-override/subscribe-mdb-changes/

The Yamcs Web UI does not currently show modified alarm or calibrator information (only modified' algorithms), that's something we'll address in a later release outside of this issue.