yamcs/yamcs #776: Cannot store negative values into a local, signed, integer parameter

merose, 2023-04-28:
To reproduce:

1. Clone Yamcs quickstart.
2. In src/main/yamcs/mdb/xtce.xml, add the SpaceSystem below just before the last line.
3. Run Yamcs quickstart via `$ mvn yamcs:run`.
4. In a browser, Open Yamcs and browse to Telemetry > Parameters, then to /myproject/Test/sampleInt.
5. Click "SET VALUE" at the top of the page.
6. In the dialog that appears, enter "-1" and press SAVE.

Expected result: New value is saved.
Actual result: Error, "Cannot fit -1 in 32 bits required for data type Int32."

[image]

Test SpaceSystem to be added to src/main/yamcs/mdb/xtce.xml just before closing `</SpaceSystem>`:

~~~
  <SpaceSystem name="Test">
    <TelemetryMetaData>
      <ParameterTypeSet>
        <IntegerParameterType name="Int32" sizeInBits="32" signed="true"/>
      </ParameterTypeSet>
      <ParameterSet>
        <Parameter parameterTypeRef="Int32" name="sampleInt">
          <ParameterProperties dataSource="local"/>
        </Parameter>
      </ParameterSet>
    </TelemetryMetaData>
  </SpaceSystem>
~~~

xpromache, 2023-04-28:
Sorry for that, fixed in master.

closed 2023-04-28 by xpromache

merose, 2023-04-28:
The method `signedSizeInBits()` in DataTypeProcessor.java appears to be the problem.

~~~
    static public final int signedSizeInBits(long v) {
        return 63 - Long.numberOfLeadingZeros(v & Long.MAX_VALUE);
    }
~~~

This will return 63 for any negative value >= 2^-62.

E.g.:
~~~
-1 & Long.MAX_VALUE --> 0x7FFFFFFFFFFFFFFF
Long.numberOfLeadingZeros(-1 & Long.MAX_VALUE) --> 1
~~~

Any negative value >= 2^-62 will have the two upper bits on, so the code above will return 63.

It's probably better to do this, although I'm not quite certain whether this results in the number of bits that matches that in the MDB for signed integers.

~~~
    static public final int signedSizeInBits(long v) {
        if (v == -1 || v == 0) {
            return 2;
        } else if (v < 0) {
            return 1 + Long.SIZE - Long.numberOfLeadingZeros(-(v + 1));
        } else {
            return 1 + Long.SIZE - Long.numberOfLeadingZeros(v);
        }
    }
~~~

You can see the failures in the existing code by adding JUnit to the pom.xml:

~~~
    <dependency>
      <groupId>junit</groupId>
      <artifactId>junit</artifactId>
      <version>4.13</version>
      <scope>test</scope>
    </dependency>
~~~

Then adding this test program at src/test/java/com/example/myproject/SizeInBitsTest.java. It sDhows values for all test values for the signed tests, but the above replacement method works.

~~~
package com.example.myproject;

import java.util.function.LongToIntFunction;
import org.junit.Test;
import org.yamcs.mdb.DataTypeProcessor;

import static org.junit.Assert.fail;

public class SizeInBitsTest {

    private static final long[][] UNSIGNED_TESTS = {
            { 1, 1 },
            { 2, 2 },
            { 255, 8 },
            { 256, 9 },
            { 0xFFFFFFFFL, 32 },
            { 0x100000000L, 33 },
            { 0xFFFFFFFFFFFFFFFFL, 64 }
    };

    private static final long[][] SIGNED_TESTS = {
            { 0x7FFFFFFFFFFFFFFFL, 64 },
            { 0x3FFFFFFFFFFFFFFFL, 63 },
            { 0x1FFFFFFFFFFFFFFFL, 62 },
            { 0x00000000000000FF, 9 },
            { 0x0000000000000003, 3 },
            { 0x0000000000000001, 2 },
            { 0x0000000000000000, 2 }, // Special case!

            { -1, 2 }, // Special case!
            { -2, 2 },
            { -3, 3 },
            { -4, 3 },
            { -127, 8 },
            { -128, 8 },
            { -129, 9 },
            { -32768, 16 },
            { -32769, 17 },
            { Long.MIN_VALUE, 64 },
    };

    @Test
    public void testUnsignedSizeInBits() {
        runTests("DataTypeProcessor unsigned", UNSIGNED_TESTS,
                v -> DataTypeProcessor.unsignedSizeInBits(v));
    }

    @Test
    public void testSignedSizeInBits() {
        runTests("DataTypeProcessor signed", SIGNED_TESTS,
                v -> DataTypeProcessor.signedSizeInBits(v));
    }

    @Test
    public void testFixedSignedSizeInBits() {
        runTests("Fixed signed", SIGNED_TESTS,
                v -> signedSizeInBits(v));
    }

    private void runTests(String caption, long[][] tests, LongToIntFunction func) {
        int failures = 0;
        for (int i=0; i < tests.length; ++i) {
            long value = tests[i][0];
            int expected = (int) tests[i][1];
            if (!checkResult(caption, expected, value, func)) {
                ++failures;
            }
        }

        if (failures > 0) {
            fail(String.format("%d failures for %s tests", failures, caption));
        }
    }

    private boolean checkResult(String caption, int expected, long v, LongToIntFunction func) {
        int actual = func.applyAsInt(v);
        if (expected == actual) {
            return true;
        } else {
            System.out.println(String.format("%s: v=%d (%016X), expected %d but was %d",
                    caption, v, v, expected, actual));
            return false;
        }
    }

    static public final int signedSizeInBits(long v) {
        if (v == -1 || v == 0) {
            return 2;
        } else if (v < 0) {
            return 1 + Long.SIZE - Long.numberOfLeadingZeros(-(v + 1));
        } else {
            return 1 + Long.SIZE - Long.numberOfLeadingZeros(v);
        }
    }

}
~~~

merose, 2023-04-28:
Sorry, wrote this long command before I saw yours. I went ahead and posted it, just in case it's helpful.

xpromache, 2023-04-30:
Thanks Mark, I will use your tests but I'm not sure about the special cases
for signed integers -1 and 0.

For me, these two are bitsize 1. I don't see why they should be special
cases with bitsize 2.

nicolae

commit f8bb2369 (2023-04-28): fixed the computation for the size in bits of signed numbers

Fixes #776