yamcs/yamcs #782: Can't Retrieve Algorithm Output Params(Through CsvParameterStreamer ) From Archive If Algorithm is Not Active After Restart.

lorenzo-gomez-windhover, 2023-06-12:
Works fine for parameter history. But not for the export CSV.

I think it comes down to this:

```org.yamcs.http.api.StreamArchiveApi.CsvParameterStreamer has a slightly different scheme than org.yamcs.http.api.ParameterArchiveApi.listParameterHistory(Context, ListParameterHistoryRequest, Observer<ListParameterHistoryResponse>) when it comes to Replay implementation. I think this might be the root of the issue.```

lorenzo-gomez-windhover, 2023-06-12:
Don't know how much this is going to help, but I went down the rabbit of hole of trying to fix this:

https://github.com/WindhoverLabs/yamcs/commit/a5c59cf34e9ebf00db33c2ba2ee5cffcb90d4206

But I figured you guys are probably going to see the problem right away.

fqqb, 2023-06-14:
CSV export performs an internal packet replay, which means re-extracting parameters from packets, and re-running algorithms. Therefore if the algorithm is no longer there, you will receive nothing.

Parameter history shows data from the Parameter Archive, which stores previously extracted parameters, hence why you do see that output data.

The solution here should be to add support to the CSV export for finding data in the Parameter Archive, and then use that on web. Will look at it when time allows.