yamcs/yamcs #783: Sample Config for TseCommander

srikarh, 2023-06-13:
Is there an example of how yamcs.yaml config is supposed to look like for the TseCommander global service? I'm getting a java.lang.ClassNotFoundException when I add it to services and on the documentation it only lists the required options.

thanks you!

fqqb, 2023-06-13:
The 'simulation' example has it configured: https://github.com/yamcs/yamcs/tree/master/examples/simulation

My guess for the ClassNotFoundException is that you didn't add yamcs-tse maven module to your project (these classes are not in yamcs-core).

srikarh, 2023-06-13:
Thank you that was it, but now I'm getting an error saying "Cannot instantiate object from class org.yamcs.xtce.TseLoader", which doesn't seem to be a class at all?

fqqb, 2023-06-13:
that was renamed to `org.yamcs.tse.TseLoader` quite some time ago.
So seems an error in the docs.

srikarh, 2023-06-14:
Got it, how would I fix this error then? I already added yamcs-tse as a dependency.
Error: Cannot instantiate object from class org.yamcs.xtce.TseLoader: java.lang.ClassNotFoundException: org.yamcs.xtce.TseLoader

srikarh, 2023-06-14:
Thank you it was my spreadsheet loader that had the wrong class.

closed 2023-06-14 by srikarh