yamcs/yamcs #785: Cannot send an empty string as a string command argument

merose, 2023-06-14:
Found in both Yamcs 5.7.x and 5.8.2.

I have created a string argument type like this:

~~~
        <StringArgumentType name="TestStringType">
          <StringDataEncoding encoding="UTF-8">
            <SizeInBits>
              <Fixed>
                <FixedValue>1024</FixedValue>
              </Fixed>
              <TerminationChar>00</TerminationChar>
            </SizeInBits>
          </StringDataEncoding>
          <SizeRangeInCharacters minInclusive="0"/>
        </StringArgumentType>
~~~

However, the Commanding > Send a Command interface will not allow me to send the command unless I have a nonempty value for that argument. It seems like setting the minimum size to zero in `SizeRangeInCharacters` should allow an empty argument. Or perhaps there is some other way of marking the argument as optional?

I suppose we could work around it by creating a custom encoding that looked for some sentinel value in the argument, like "-none-" or "\0" and converted that to an empty string, but it seems like empty string arguments should be allowed.

fqqb, 2023-06-15:
Agree, empty should be allowed. I'd say also if SizeRangeInCharacters is unspecified.
Looking at it, it seems currently we don't even read the SizeRangeInCharacters, will fix.

fqqb, 2023-07-05:
On master, for inclusion in next release.

closed 2023-07-05 by fqqb