yamcs/yamcs #787: Yamcs 5.8.x cannot run behind reverse proxy HTTP server

merose, 2023-06-14:
Prior to 5.8, it was possible to run Yamcs behind a reverse proxy HTTP server as follows:

- Create an etc/security.yaml file.
- Set `enabled: false` in etc/security.yaml, to allow requests that bypass the reverse proxy server to use the local user account.
- Configure the reverse proxy to use HTTP basic authentication and forward the "Authorization" header.
- Create users as needed, with dummy passwords.

However, with the change to Yamcs 5.8, this no longer works. The reason is that the check for the "Authorization" header is not done if security is disabled through `enabled: false` in etc/security.yaml.

In 5.7.8, yamcs-core/src/main/java/org/yamcs/http/HttpRequestHandler.java, lines 236-246:

~~~
        // Handle common case first: presence of an "Authorization" header
        if (req.headers().contains(AUTHORIZATION)) {
            String authorizationHeader = req.headers().get(AUTHORIZATION);
            if (authorizationHeader.startsWith(AUTH_TYPE_BASIC)) { // Exact case only
                return handleBasicAuth(ctx, req);
            } else if (authorizationHeader.startsWith(AUTH_TYPE_BEARER)) {
                return handleBearerAuth(ctx, req);
            } else {
                throw new BadRequestException("Unsupported Authorization header '" + authorizationHeader + "'");
            }
~~~

However, in 5.8.2, yamcs-core/src/main/java/org/yamcs/http/HttpHandler.java, lines 69-81, the check for a basic authentication header is not done unless the securityStore is enabled:

~~~
        var securityStore = YamcsServer.getServer().getSecurityStore();
        if (securityStore.isEnabled()) {
            // Handle common case first: presence of an "Authorization" header
            if (req.headers().contains(AUTHORIZATION)) {
                String authorizationHeader = req.headers().get(AUTHORIZATION);
                if (authorizationHeader.startsWith(AUTH_TYPE_BASIC)) { // Exact case only
                    return handleBasicAuth(ctx, req);
                } else if (authorizationHeader.startsWith(AUTH_TYPE_BEARER)) {
                    return handleBearerAuth(ctx, req);
                } else {
                    throw new BadRequestException("Unsupported Authorization header '" + authorizationHeader + "'");
                }
            }
~~~

With the securityStore enabled, the web UI tries to do the token authentication using the login page, even though Yamcs already has an Authorization header sent on every request.

I realize that the current server manual says that HTTP basic authentication doesn't work with the Yamcs web UI, but it used to. We'll have to work around this by modifying HttpHandler.java, I think. We want to use single-sign-on for several different HTTP services that work with Yamcs, so letting the Apache server handle authentication is very important. And we want the web UI to work and not try to do the token authentication.

fqqb, 2023-06-15:
This change occurred as of 5.7.9: https://github.com/yamcs/yamcs/commit/ea89c8e6a405ba4c4acb487cdecaf13cc9c27d73

Instead of passing an Authorization header, can you pass a custom header with the desired username as value?
Then add this to your security.yaml:

```yaml
enabled: true
authModules:
  - class: org.yamcs.security.RemoteUserAuthModule
    args:
      header: "X-REMOTE-USER"
```

Yamcs will automatically create the user if it does not yet exist.

fqqb, 2023-06-16:
Upon further reflection, my previous answer does not help, because as soon as security is enabled, the webapp will still redirect to the login form, which is exactly what you don't want.

Thinking how best to resolve this, either a yamcs-web configuration property to disable this redirect, or maybe a way to automatically convert to the token flow (essentially bypassing the login form). Not sure yet.

merose, 2023-06-30:
Note that this is related to #793, which is another issue found when running Yamcs behind a reverse proxy server.

fqqb, 2023-07-05:
On master, for inclusion in next release.

I added a new property `disableLoginForm` that you can add to yamcs-web configuration section in `etc/yamcs.yaml`:

```yaml
yamcs-web:
  disableLoginForm: true
```

It has the effect of not redirecting to the login page when no token is detected. If auth is done through some other mechanism (basic, custom), the webpage will still work.

closed 2023-07-05 by fqqb