yamcs/yamcs #788: Yamcs with basic authentication has concurrency bug causing privilege loss

merose, 2023-06-15:
I have not yet been able to reproduce this using Yamcs quickstart, but can reproduce it using our configuration.

Reproduced  in both Yamcs 5.7.8 and 5.8.2.

Here's the setup:

- In Yamcs 5.8.2, enable security in etc/security.yaml.
- Put an HTTP reverse proxy in front of Yamcs that uses basic authentication and forwards the Authentication header.
- Create a new user in the Yamcs admin area.

Here's the problem:

- If multiple threads in clients are making requests simultaneously, the basic authentication cases Yamcs to reload the user's privileges in the User object for each request. But the access to User is not properly synchronized, so some threads may see a User object with privileges unpopulated.
- Example scenario: 100 threads making requests to read the current value of a parameter once per second, running for 10 seconds. About 2% of the requests (on my machine) get 403 errors in response, "unauthorized."

I've worked around the problem by fixing Directory#updateUserProperties() to synchronized on the User, and by synchronizing all the getters/setters in User as well. Here's the code in 5.8.2 in Directory:

~~~
    public synchronized void updateUserProperties(User user) throws IOException {

        // Recalculate effective privileges
        user.clearDirectoryPrivileges();
        for (String roleName : user.getRoles()) {
            Role role = getRole(roleName);
            if (role != null) {
                for (SystemPrivilege privilege : role.getSystemPrivileges()) {
                    user.addSystemPrivilege(privilege, false);
                }
                for (ObjectPrivilege privilege : role.getObjectPrivileges()) {
                    user.addObjectPrivilege(privilege, false);
                }
            }
        }

        users.put(user.getName(), user);
        persistChanges();
    }
~~~

As you can see, first `user.clearDirectoryPrivileges()` is called, then the privileges are filled in. Because this whole operation is not atomic, some other thread may see the User object after `clearDirectoryPrivileges()` but before the privileges are added back. That other thread would then get a 403 error, because the user had no privileges appropriate for reading any parameters.

As I said, I have not been able to reproduce the problem using Yamcs quickstart but am still working on it. That said, I think it's a legitimate problem. I'm attaching a patch that fixes the problem in my setup.

[patch.txt](https://github.com/yamcs/yamcs/files/11751496/patch.txt)

fqqb, 2023-06-15:
Yes, I can see how that could happen. Will investigate.

fqqb, 2023-07-28:
I was able to reproduce this reliably by artificially delaying responses.

This should now be fixed on master through the use of a user cache. Please give it a try, and let us know if you still find issues.

closed 2023-07-28 by fqqb