yamcs/yamcs #789: Question about TSE Commanding

srikarh, 2023-06-21:
I currently have TSE commander setup to take in commands with configuration set up as well. I'm sending commands from yamcs studio and yamcs is able to capture the command but the command is not being sent out to the test equipment. I also tested the telnet interface of the TSE commander and it works but I'm not sure how the commands can be sent out. I'm not sure if I need to set up a datalink or if something is wrong with my config. Thank you!

fqqb, 2023-06-22:
Do you have a link of type `org.yamcs.tse.TseDataLink` ? Is its out-counter increasing when you send a command? Is there any output in logs.

srikarh, 2023-06-22:
I do have that data link, for some reason the udp-out count is going up while the tse out-count isn't.

fqqb, 2023-06-23:
TseDataLink expects to receive command instructions on a stream named tc_tse, to distinguish from regular non-tse commands.

You could use something like this in the `streamConfig` of your instance configuration:

```yaml
streamConfig:
  tc: 
     - name: tc_tse
       processor: realtime
       tcPatterns: ["/TSE/.*"]
     - name: tc_realtime
       processor: realtime
```

With the idea that commands in the namespace /TSE are emitted on tc_tse stream (for use by TseDataLink), and other on the standard tc_realtime stream (for your udp-out).

srikarh, 2023-06-23:
Thank you was able to figure it out!

closed 2023-06-23 by srikarh