yamcs/yamcs #790: TcpTmDataLink Invalid packet read: packetLength (2823) > maxPacketLength(1500)

srikarh, 2023-06-26:
I'm not sure why I'm getting this error as my packets should be 144 bytes and I've confirmed it on Wireshark, is there a setting I need to change to allow this datalink to pick up the packets?

xpromache, 2023-06-26:
Can you look in the wireshark captures where the 2823=0x0B07 appears and if your 144=0x90 appears in some other place?

xpromache, 2023-06-26:
Sorry, not sure what your question is. 
What exactly is a frame and how does it relate to the TCP packet?

The size of TCP packets is controlled by the OS (because TCP works as a stream really)

srikarh, 2023-06-26:
Sorry for the confusion I have packet of size 144 bytes but on Wireshark I can see that the frame is 210 bytes, I don't see 2823 anywhere. I was wondering if the error was with yamcs not allowing bigger packet sizes and if I could configure that?

xpromache, 2023-06-26:
Your TCP stream has to have some delimiters inside because otherwise Yamcs cannot know when a packet starts and when it ends.

Typically in front of each packet you would have 2 or 4 bytes which indicate the packet length. Or for CCSDS you have the 5th and 6th bytes indicating the length.

If your packets are in fact 144 bytes it is definitely because either you do not have that length inside the stream or Yamcs looks for it at the wrong place. Increasing the size (above the default 1500) won't solve your problem since your packets are smaller than that.

So what kind of packets do you have and how are they encoded into the stream?

srikarh, 2023-06-26:
Thanks for the information I'm sending a TCP packet of 144 bytes and right now I don't have any headers and I'm not using CCSDS or delimiters. I'm just sending pure binary packets of 144 bytes. What do you recommend adding in terms of delimiters and headers and is there an example that I could look at? How would you configure yamcs to look for the info in the header and would you be able to provide an example of receiving non-CCSDS binary data? thank you!

xpromache, 2023-06-27:
If all your packets are exactly the same length, try this config:
dataLinks:
...
    - name: tm_realtime
     class: org.yamcs.tctm.TcpTmDataLink
   ....
     packetInputStreamClassName: *org.yamcs.tctm.FixedPacketInputStream*
          packetInputStreamArgs:
              *packetSize: 144*

srikarh, 2023-06-27:
Thank you for the example, I'm able to get data now but the hex is flipped how would you be able to change the endianness of parameters in Yamcs? thanks!

xpromache, 2023-06-28:
> 
You can do that in the MDB definition. How do you define your MDB?

srikarh, 2023-06-28:
Gotcha I'm using XML to define my MDB

srikarh, 2023-07-20:
Just to follow up how would I change the endianness in the xml mdb?

fqqb, 2023-07-24:
There's a byteOrder attribute on the encoding, example: https://github.com/yamcs/yamcs/blob/f3ba1534c38b3990a6cdb5a1b3c2ec302af97a0b/yamcs-core/src/test/resources/xtce/BogusSAT-2.xml#L332

This is specified in the XML schema: https://www.omgwiki.org/space/doku.php?id=xtce

closed 2024-03-06 by fqqb