yamcs/yamcs #791: Error when setting maxSize of bucket to 5GB

AdamElliott-AB, 2023-06-27:
In yamcs.yaml:

```
  - name: bucket1
    path: bucket1
    maxSize: 5368709120 # 5 GB
    maxObjects: 5000
```

Error on Yamcs start:

```
15:40:48.540 _global [1] YamcsServer Failure while attempting to validate configuration: java.lang.IllegalArgumentException: Cannot derive type for argument of class java.lang.Long
	org.yamcs.Spec$OptionType.forArgument(Spec.java:430)
	org.yamcs.Spec$OptionType.convertArgument(Spec.java:391)
	org.yamcs.Spec$Option.validate(Spec.java:661)
	org.yamcs.Spec.doValidate(Spec.java:240)
	org.yamcs.Spec$Option.validate(Spec.java:680)
	org.yamcs.Spec.doValidate(Spec.java:240)
	org.yamcs.Spec.validate(Spec.java:149)
	org.yamcs.YamcsServer.validateMainConfiguration(YamcsServer.java:1199)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1154)
	org.yamcs.YamcsServer.main(YamcsServer.java:959)
```

fqqb, 2023-06-28:
Right, shall be fixed in next release.

fqqb, 2023-07-05:
On master, for inclusion in next release.

closed 2023-07-05 by fqqb