yamcs/yamcs #793: Yamcs 5.8 web client embeding encoded slashes in URL paths will not work with Apache reverse proxy

merose, 2023-06-29:
Some Yamcs 5.8 web UI requests embed slashes into the path portion of the URL. When fronting Yamcs with a web proxy server such as Apache this can cause problems, since Apache does not allow encoded slashes (%2F) in the URL.

For example, browsing to Mission Database > Commands, and then a command name results in this type of error in the Apache error log:

[Thu Jun 29 14:06:22.826833 2023] [core:info] [pid 255082:tid 139774157362944] [client ::1:60236] AH00026: found %2f (encoded '/') in URI path (/yamcs/api/mdb/viper/commands/%2FCcsds%2FCommandHeader), returning 404, referer: http://localhost:8080/yamcs/mdb/commands/%2FCcsds%2FCommandHeader?c=viper__realtime

## Workaround

You can add `AllowEncodedSlashes On` to the Apache configuration and Yamcs then operates properly. (The default is `Off`.) However, the Apache documentation recommends against this for security reasons. In addition, using `NoDecode` instead of `On`, which is recommended if the default will not work, does not solve the problem.

## Discussion

It appears that only some requests coming from the Yamcs web client encoded XTCE paths in this way in the URL. Others, such as used by parameter access in Telemetry > Parameters, use the slashes in the XTCE path unencoded in the URL, and work fine with an Apache reverse proxy.

fqqb, 2023-07-03:
I greatly dislike the %2F escapes myself, but it removes any path ambiguity issues that could otherwise happen. It's an API design problem really, which we should be addressing.

> However, the Apache documentation recommends against this for security reasons.

I've seen this. They don't specify what exactly the security issue is. So I'm a bit skeptical about that.

> It appears that only some requests coming from the Yamcs web client encoded XTCE paths in this way in the URL. Others, such as used by parameter access in Telemetry > Parameters, use the slashes in the XTCE path unencoded in the URL, and work fine with an Apache reverse proxy.

When you navigate within the web application, Apache is not involved (client-side routing). Only when you F5 you may see potential problems.

duane-impulse, 2023-07-25:
I ran into this as well, and the same solution mentioned in the original issue `AllowEncodedSlashes On` worked.

> When you navigate within the web application, Apache is not involved (client-side routing). Only when you F5 you may see potential problems.

Maybe I misunderstand you, but this was not the case for me. The send command page and attempting to view individual commands in the mdb browser weren't loading properly when navigating using only clicks on the web interface. As stated in the original comment, all of the telemetry browsing worked nominally.

fqqb, 2024-01-28:
Getting good results by using `AllowEncodedSlashes NoDecode`, if combined with the `nocanon` keyword on the `ProxyPass`.

Without the `nocanon` the `%2F` arrives doubly-escaped as `%252F`, so it is needed.

Some interesting background can be found here: https://bz.apache.org/bugzilla/show_bug.cgi?id=35256