yamcs/yamcs #796: Binary argument types do not display properly in the 5.8.2 web UI

merose, 2023-06-30:
To reproduce:

1. Clone the Yamcs quickstart project.
2. In the quickstart project, add the SpaceSystem below to the bottom of src/main/yamcs/mdb/xtce.xml.
3. Start yamcs via `mvn yamcs:run`.
4. Log in to the Yamcs web UI.
5. Go to Commanding > Send a Command, then browse to the command /myproject/Test/Sample.

Expected result: The single command argument `binary` has a form field for typing in a value.
Actual result: The "binary" label for the command argument appears, but no form field, making it impossible to send the command.

[image]

The additional SpaceSystem to add to the bottom of src/main/yamcs/mdb/xtce.xml:

~~~
  <SpaceSystem name="Test">
    <CommandMetaData>
      <ArgumentTypeSet>
        <BinaryArgumentType name="Binary">
          <BinaryDataEncoding>
            <SizeInBits><FixedValue>16</FixedValue></SizeInBits>
          </BinaryDataEncoding>
        </BinaryArgumentType>
      </ArgumentTypeSet>
      <MetaCommandSet>
        <MetaCommand name="Sample">
          <ArgumentList>
            <Argument argumentTypeRef="Binary" name="binary"/>
          </ArgumentList>
          <CommandContainer name="SampleContainer">
            <EntryList>
              <ArgumentRefEntry argumentRef="binary"/>
            </EntryList>
          </CommandContainer>
        </MetaCommand>
      </MetaCommandSet>
    </CommandMetaData>
  </SpaceSystem>
~~~

## Workaround

You can still send a command with a binary argument using a Python script or a curl command. It appears that the problem is limited to the Yamcs web UI.

fqqb, 2023-06-30:
Yes, I've seen this. It's another sign we should start adding some unit tests to yamcs-web for these basic functionalities.

Fix has been prepared and will be in my next PR. I want to include few other things too before then.

fqqb, 2023-07-05:
On master, for inclusion in next release.

closed 2023-07-05 by fqqb