yamcs/yamcs #797: Would prefer continued support of XML command stack format

merose, 2023-06-30:
We have created an infrastructure that allows creation of binary relative time sequence and absolute time sequence commands from XML-based command stacks. While the new JSON format is undoubtedly more feature-rich, we have limited resources available to convert our infrastructure to the new format. It would be helpful if the older, XML format is still supported, even if deprecated, until some time in 2024. We're probably going to finalize our Yamcs version in early 2024. We don't need any of the new features supported in the XML format.

merose, 2023-06-30:
After talking with our flight-software team, it would be convenient to preserve the option of saving in the old format, too, though we may be able to work around that with a conversion script.

fqqb, 2023-07-03:
Should be fine, no plans to remove the XML option anytime soon, especially if we know you use it.

The additional functions in the new format were added at your request (#743), so that part is a little surprising, but OK.

merose, 2023-07-03:
Yes, sorry to be unclear. We love the new features in the new, JSON format, and we are using them already in stacks created for our operations. At the same time, our flight software team created infrastructure to create the binary format needed for cFS stored commands by converting from the XML stack format. They may be able to convert to using the new format, but for the time being were worried that a Yamcs change removing the XML format could break their processes for creating stored command tables. So for now we're using both formats for different purposes.