yamcs/yamcs #798: How do you hide the left menu in the 5.8.2 Yamcs UI?

merose, 2023-06-30:
In prior Yamcs versions you could hide the left menu by pressing a button in the web UI. That feature appears to have been removed. In a mission control environment, when screen area is very precious because of the amount of displays and applications that you have to arrange, being able to hide the left menu is very important.

How can we enable that feature for Yamcs 5.8.2?

fqqb, 2023-07-03:
I removed the side toggle because I felt there was too much clutter in the top-left (still is actually) for a functionality I was not entirely happy with to begin with.

I agree on your rationale. I'm thinking this use case could also be addressed by an option that 'expands the inner page' rather than 'hides the sidebar'. Then it could hide both the top and the side. Even a fullscreen option (although I don't think that will help you specifically).

fqqb, 2023-07-28:
Added an option "Toggle Zen Mode" in the top-right menu (next to mission time).

It has the effect of hiding both the sidebar and the top menu. The change is not persisted, so a refresh will get rid of the effect (or manually untoggle in the same menu).