yamcs/yamcs #800: Any Plans to Decouple WebSocketClient Class from Browser Environment?

lorenzo-gomez-windhover, 2023-07-05:
First off, thanks a lot for decoupling the "sdk" front-end code from all of the angular!

https://github.com/yamcs/yamcs/blob/f3ba1534c38b3990a6cdb5a1b3c2ec302af97a0b/yamcs-web/src/main/webapp/projects/webapp-sdk/src/lib/client/WebSocketClient.ts#L32

The reason I bring this up is because if someone is using the webapp-sdk(we are for instance) as a dependency in nodejs as opposed to the browser, then it will not compile(worse yet; will throw some undefined exception at runtime).

It's not that big of a deal since there is a workaround (at least one that I can think of):

- extend the YamcsClient and override the behavior of `prepareWebSocketClient`

However, it'd be nice to be able to use it on the server side.

Thanks
Lorenzo

lorenzo-gomez-windhover, 2023-07-05:
I guess another possible solution would be adding "window" to the `global` dictionary in Node....

fqqb, 2023-07-05:
You may already have derived, but I add some context here for the record: the primary motivation for this "webapp-sdk" is a work-in-progress plan to allow at-runtime addition of mission-specific pages (that is, without recompiling the core Yamcs webapp). This first iteration extracts the client code, and also a bunch of the base components. Future iterations will work on how to handle auth/communication between the main app and one of those extension pages.

Making a general JavaScript client would also be nice, and something that has been on my mind for a while. The current SDK is not published to npm, so I wasn't expecting anybody to pick up on it *yet* (ideally client portion should be another @yamcs/client package or something, which then the SDK depends on), but I'll see what I can do to make it node-friendly.

lorenzo-gomez-windhover, 2023-07-05:
Thanks for the detailed clarification! It makes a lot of sense to do this given how extensible the YAMCS server is via Java plugins.

lorenzo-gomez-windhover, 2023-07-05:
Posting my workaround for anyone else who runs into the same  issue:

```ts
        // TODO:I'm not sure if this is the best way to do this...
        // Another option would be to extend classes and override methods
        globalThis.window.location = new Location;
        globalThis.window.location.hostname = yamcsServerIP;
        let client = new YamcsClient(yamcsServerIP, new FrameListener());
```

closed 2023-07-05 by lorenzo-gomez-windhover