yamcs/yamcs #803: Potential bug in 5.8.3 - Unknown argument host

nmaas87, 2023-07-10:
Dear all,
I just updated from Yamcs 5.8.1 to .3 - Yamcs does crash on launch with errors in the UdpTmDataLink:

entry from config:
````
dataLinks:
  - name: udp-in
    enabledAtStartup: true
    class: org.yamcs.tctm.UdpTmDataLink
    stream: tm_realtime
    host: 10.0.0.1
    port: 10000
    packetPreprocessorClassName: com.example.TEST.TESTPreprocessor
    packetPreprocessorArgs:
      performCRCCheck: true

````

log:
````
Jul 10 07:16:42.125 TEST [17] org.yamcs.yarch.TableReaderStream[memento_1] [FINE] Closed stream memento_1 num emitted tuples: 0
Jul 10 07:16:42.125 TEST [17] org.yamcs.yarch.SelectStream[memento_1_select] [FINE] Closed stream memento_1_select num emitted tuples: 0
Jul 10 07:16:42.142 TEST [1] org.yamcs.management.LinkManager [FINE] Raw args for udp-in: {name=udp-in, enabledAtStartup=true, class=org.yamcs.tctm.UdpTmDataLink, stream=tm_realtime, host=10.0.0.1, port=10000, packetPreprocessorClassName=com.example.TEST.TESTPreprocessor, packetPreprocessorArgs={debug=true}}
Jul 10 07:16:42.143 _global [1] org.yamcs.YamcsServer [SEVERE] yamcs.TEST.yaml: dataLinks[0]: Unknown argument host
Jul 10 07:16:42.144 _global [1] org.yamcs.yarch.rocksdb.RDBFactory [FINE] Shutting down. Closing 1 databases under /storage/yamcs-data/_global.rdb: []
Jul 10 07:16:42.145 _global [1] org.yamcs.yarch.rocksdb.RDBFactory [FINE] Shutting down. Closing 1 databases under /storage/yamcs-data/TEST.rdb: []
````

Is this a bug or am I the problem?

Regards,
Nico

fqqb, 2023-07-10:
Not a bug, there's no such option: https://docs.yamcs.org/yamcs-server-manual/links/udp-tm-data-link/
We have it on some other link types.

nmaas87, 2023-07-10:
Interesting, because in 5.8.1 this did still work (I have been using it in OPS since end of 2019). Do I just need to rename host to stream or what was the change here?
Thanks
Nico

fqqb, 2023-07-10:
Because this wasn't validated before.

nmaas87, 2023-07-10:
Ah I see, so it basically never worked and always used all the servers interface IPs to receive Udp TM on all interfaces, not the one specified by a user, correct?

fqqb, 2023-07-10:
Correct. Can be left open to implement though.