yamcs/yamcs #804: Webmanifest file is rejected by apache when used as a reverse proxy

unlikelyzero, 2023-07-10:
At the moment, we're seeing 401s from apache when requesting the webmanifest file (without modifications to our apache config).

I believe that the issue can be corrected with:
https://medium.com/@aurelien.delogu/401-error-on-a-webmanifest-file-cb9e3678b9f3

https://github.com/yamcs/yamcs/blob/f3ba1534c38b3990a6cdb5a1b3c2ec302af97a0b/yamcs-web/src/main/webapp/projects/webapp/src/index.html#L15
```
<!--link rel="manifest" href="{{ contextPath }}/manifest.webmanifest"-->
```

Should be
```
<link rel="manifest"  href="{{ contextPath }}/manifest.webmanifest" crossorigin="use-credentials">
```

fqqb, 2023-07-11:
Interesting, I'll add it.
The 401 would come from your Apache-level auth interception is my assumption, because Yamcs itself does not protect this resource.

I also think you're currently using Yamcs 5.8.2. As of 5.8.3 that line is commented out, so I don't expect there to even be a request for it at all.

Some more detail: in Yamcs 5.8.2 I enabled for the first time some PWA behaviours via the webmanifest (allows browser to download full web application, which has some interesting potential use cases). It seemed OK for a while, but then we started having some issues with unintended caching of paths under /api and /auth. Therefore the functionality is disabled in 5.8.3, until another time. https://github.com/yamcs/yamcs/commit/a47e293f066c83e869b9e1892e9a9edae85a7b02

fqqb, 2023-07-28:
Added the crossorigin attribute on master.
The tag will remain commented out for the time being.

closed 2023-07-28 by fqqb