yamcs/yamcs #821: When there are multiple root containers, "extract" in packet list does not extract correctly

merose, 2023-08-03:
I don't have a detailed description yet, but working on it.

To reproduce:

1. Configure MDB to have at least two root containers.
2. Configure two data links to have separate root containers.
3. Send one packet to each data link that have different root containers.
4. In the Yamcs Home page, verify that both packets have been received.
5. In Telemetry > Packets, click the Extract button for each packet.

Expected result: Both packets extract correctly.

Actual result: Only one root container is used, causing incorrect extraction of one of the packets.

fqqb, 2023-09-04:
Addressed in PR #827

This function will now attempt to find a suitable root container, based on the stored packet name and the current MDB.

fqqb, 2023-09-19:
On master, and in latest release.

closed 2023-09-19 by fqqb