yamcs/yamcs #822: Having multiple different sources for the same parameters

Hixos, 2023-08-04:
Hi everyone, at my company we are currently evaluating whether YAMCS could fit our groundstation need.

In my setup, I have multiple identical systems generating telemetry.
To avoid code duplication (in the XTCE), my plan was to define the telemetry parameters only once, reusing them in every system, together with an enumerated parameter that defines from which source the specific parameter value is originating from.  
  
After looking for a while though, I could not find a way to display values from different sources separately in Yamcs web / Studio, since everything is being stored in the same parameter. 

### Minimal example:
Let's say I have two identical temperature sensors. I define the XTCE as follows:
```xml
<xtce:TelemetryMetaData>
    <xtce:ParameterTypeSet>
        <xtce:EnumeratedParameterType name="TemperatureSensorEnumType">
            <xtce:EnumerationList>
                <xtce:Enumeration value="0" label="Sensor1" />
                <xtce:Enumeration value="1" label="Sensor2" />
            </xtce:EnumerationList>
        </xtce:EnumeratedParameterType>
        <!-- ... -->
    </xtce:ParameterTypeSet>

    <xtce:ParameterSet>
        <xtce:Parameter parameterTypeRef="TemperatureSensorEnumType" name="SensorID" />
        <xtce:Parameter parameterTypeRef="TemperatureType" name="Temperature" />
    </xtce:ParameterSet>

    <xtce:ContainerSet>
        <xtce:SequenceContainer name="TemperatureSensorData">
            <xtce:EntryList>
                <xtce:ParameterRefEntry parameterRef="SensorID" />
                <xtce:ParameterRefEntry parameterRef="Temperature" />
            </xtce:EntryList>
        </xtce:SequenceContainer>
    </xtce:ContainerSet>

</xtce:TelemetryMetaData>
```

The two systems are transmitting the same `TemperatureSensorData` container, only with a different `SensorID`.
What would be the best approach to then be able to treat the `Temperature` value separately based on the source? An example would be plotting the two temperatures in two different traces.  

### Possible ideas
- On https://github.com/yamcs/yamcs/issues/303#issuecomment-767647287, it is suggested to have two different root spacesystems for the same database. I guess this could be adapted to my problem by somehow filtering packets from the two sources to one spacesystem or the other.  
- Creating local parameters and assigning them based on the `SensorID`, but this would still require lots of boilerplate code.

I'm not sure what the best way forward is here, so I'm open to suggestions.

fqqb, 2023-08-04:
If you want to have distinct parameters, they should somehow have distinct names. Either at path level, or the actual parameter name if you prefer to keep it flat.

From your description it sounds like you have different packets that fully look the same, but just come from different systems.

To me that sounds like a good use case for nested space systems. By that I mean you would map to different parameters `/myproject/sub1/Temperature` and `/myproject/sub2/Temperature`, sub1 and sub2 being the different systems (nested `SpaceSystem` elements in XTCE).

To distinguish between the sub1 container and the sub2 container based on an id in some header, you could use `BaseContainer`/`RestrictionCriteria`.

Or indeed the rootContainer concept, then you don't really need to distinguish them based on bit pattern, as it will already be clear at link level.

Tagging @xpromache as he may have better suggestions.

Hixos, 2023-08-04:
Thanks for you prompt reply!   
The nested spacesystem solution is something that I thought about as well, but it would not be ideal because in our real use case as we would have to repeat hundreds of parameters over multiple identical spacesystems, making it very difficult to maintain.  
I guess automating this process would not be too difficult, but before getting into this endeavor I wanted to know if there are any better options available.

fqqb, 2023-08-04:
Containers can also include entries which are another container (`ContainerRefEntry`).

In other words, you could have some shared container/parameter definition in your root space system, which you then reference in each of your systems. The system-specific container is then very short, just the ID, and a reference to some shared other container definition.

Agree on the automation bit, I'd say the XTCE XML is better generated than hand-maintained.

fqqb, 2023-08-04:
Hmm having second thoughts. Need to think this more through.

For sure at least the parametertypes could be shared.

fqqb, 2023-08-04:
What I said before with the `ContainerRefEntry` would not work because the parameter names would be identical, which is exactly what you don't want.

I believe shared parametertypes is the most you can do in terms of reuse within the XML.

fqqb, 2023-08-07:
One more comment here, there is a way to increase reuse by splitting systems in different XTCE files. Then instead of doing the nesting inside the XML, specify the hierarchy in YAML.

Something like this:

```yaml
mdb:
  - type: xtce
    args:
      file: root.xtce
    subLoaders:
      - type: xtce
        args:
          file: system1.xtce
        subLoaders:
          - type: xtce
            args:
              file: subsystem1.xtce
          - type: xtce
            args:
              file: subsystem2.xtce
      - type: xtce
        args:
          file: system2.xtce
        subLoaders:
          - type: xtce
            args:
              file: subsystem1.xtce
          - type: xtce
            args:
              file: subsystem2.xtce
```

In this example `subystem1.xtce` and `subsystem2.xtce` are shared between both `system1` and `system2`, yet their content will have different parameter names based on where in the tree it is included.

Hixos, 2023-08-07:
This last idea is actually very nice and fits my original intentions quite well.
 
With this, I think I can close this issue, thanks a lot for your help!

closed 2023-08-07 by Hixos