yamcs/yamcs #824: The samples endpoint returns NaN for enums

akhenry, 2023-08-21:
The "samples" endpoint returns results for enum parameters with timestamps, but all of the numerical fields contain `NaN`. I'm unsure if this is by design, or if it is a bug.

It would be extremely useful to our users to support samples of enum parameters, based on the integer value. Clearly an "average" doesn't make sense, but a min and max would be useful for our purposes. Users of Open MCT commonly build plots of enum parameters that look like the example below, and having to populate these with an exhaustive query of the parameter archive is extremely slow.

[image]

Reproduction steps:
1. Clone the Yamcs Quickstart repository and build per the instructions
2. Start the Yamcs Quickstart server
3. Start the Python simulator provided in the Yamcs Quickstart repository
4. Navigate to http://localhost:8090/telemetry/parameters/%2Fmyproject%2FEnum_Para_1/chart?c=myproject__realtime&interval=PT15M
5. Observe that the plot is empty (see screenshot below)
6. Observe that a request has been issued to the `samples` endpoint, and that the corresponding HTTP response is all `NaN` values. (see screenshot below)
[image]

AdamElliott-AB, 2023-08-22:
I second @akhenry 's request. It is extremely useful to view an Enum's value history in the chart view to see when value changes occurred, even if the chart is only showing the enum's raw values. It would be easy to support this by modifying Downsampler.java and DyDataSource.ts to fallback to using the raw values if a given parameter's eng type is non-numeric but it's raw type is numeric.

fqqb, 2023-08-22:
Enumerations are more similar to a string than to a number (same for boolean), hence why the have always been excluded from the sampler and plots. I agree the min/max may be useful regardless, so that's something we could do.

The sampling API does already provide a ``useRawValue`` query parameter option, which then returns samples on the raw (numeric) value. It tried it with the quickstart project, and it works there, so that may be a solution for the OpenMCT visualization.

Then for yamcs-web, maybe anyway we add a function to allow showing the raw on the plot, but then for all types of parameters. Longer-term desired solution for yamcs-web is to have a custom visualization that makes use of the Ranges API (which is specifically intended to query discrete parameters).

AdamElliott-AB, 2023-08-22:
I like the idea of the yamcs-web chart option for a user to select to view rawValues. Even so, it makes sense to me to automatically fallback to the rawValue option if the engValue is not chartable.

akhenry, 2023-08-22:
@fqqb @xpromache These are both excellent solutions, thank you. `useRawValue` works great and will tide us over in the short term. 

Ranges seem like the better solution in the long term so we will look to migrating our enum parameter archive requests to use ranges.

closed 2023-08-22 by akhenry

fqqb, 2023-08-22:
Cfr here for an example on how to use the ranges API (it's not an easy method):
https://github.com/yamcs/grafana-yamcs/blob/master/src/DataSource.ts#L271