yamcs/yamcs #825: Yamcs web window title overlaps with header navigation when window is narrow

merose, 2023-08-31:
In a mission operations center environment, some users create multiple browser windows, some of which can be quite narrow to show only certain information. In these cases the Yamcs window title in the header bar can overlap with buttons to its left and right. (In addition, we tend to put additional information in the title using "yamcs-web: tag:", which makes the problem worse for us.

We have found this CSS snipped, added via "yamcs-web: extraHeaderHTML:", works well to reduce the font size, make the navigation areas opaque, and put the title into the lowest Z position. Perhaps consider something similar for addition into the standard CSS.

~~~
yamcs-web:
  # Put title bar behind buttons in the header, so that the window can
  # be narrowed without having the window title overlap with buttons.
  extraHeaderHTML: "<style>
.app-bar div.tag {
    font-size: 10pt;
    z-index: 0;
}
.app-bar .mat-tab-link, .app-bar button {
    background-color: rgb(33,33,33);
}
.logo-wrapper {
    z-index: 1;
    background-color: rgb(33,33,33);
}
</style>"
~~~

fqqb, 2023-09-04:
Addressed in PR #827.
Lowered font size, and added `overflow: hidden` to the tag div, it seems to work identical to your example.

Probably we can/should be better for small screens in general, but that's for another time.

Note in your example, it should be `.mat-mdc-tab-link` instead of `.mat-tab-link` (classname changed during a recent Angular upgrade)

fqqb, 2023-09-19:
On master, and in latest release.
Altough I did make the 'tag' a bit larger (14px) because it was looking out of place at 10px. Still smaller than the previous 20px.

Next I'm maybe going to reduce the size of the top right buttons, because they are looking huge to me now

closed 2023-09-19 by fqqb