yamcs/yamcs #831: Performance regression with authorization

akhenry, 2023-09-26:
Since we started using version 5.8.3 of Yamcs we have noticed that all requests to Yamcs API services take much longer to complete. This also affects the Yamcs Web UI, which is noticeably slower.

It took some time to isolate the issue but it appears to be related to authorization. Only authorized requests are delayed, and response times increase in direct proportion to the size of the user database.

The issue is not present in Yamcs version 5.7.8.

I have produced [a minimal reproduction](https://github.com/yamcs/quickstart/compare/master...akhenry:quickstart:auth-perf-regression) of the issue using Yamcs Quickstart project with instructions below.

### Reproduction Steps
#### 1. Clone reproduction fork of the Yamcs Quickstart project
```
git clone -b auth-perf-regression git@github.com:akhenry/quickstart.git akhenry-quickstart
cd akhenry-quickstart
```

#### 2. Compile and run Yamcs Quickstart
```
mvn compile
mvn yamcs:run
```

#### 3. Issue a simple request to the Yamcs HTTP API
```
% curl -u user:password -o /dev/null -s -w 'Response Code: %{response_code} Time: %{time_total}s\n' http://localhost:8090/api/mdb/myproject
Response Code: 200 Time: 0.061049s
```
Note that the response time is quick, ~60ms in this case.

#### 4. Load 150 users into Yamcs
Using the provided script, load 150 users into the database. This is the approximate total number of users we will need to track for our mission.
```
node/node load-users.js 150
```
You will note how long it takes just to create the users and assign a role to them.

#### 5. Issue the same request to the Yamcs HTTP API
```
curl -u user:password -o /dev/null -s -w 'Response Code: %{response_code} Time: %{time_total}s\n' http://localhost:8090/api/mdb/myproject
Response Code: 200 Time: 0.971510s
```
Note that the response time has increased to ~1s. This will be the case for any authorized requests to Yamcs.

#### 6. Repeat steps 2-5 using Yamcs version 5.7.8
Note that when you get to step 5, the response times are much quicker, even with 150 users loaded. eg.
```
curl -u user:password -o /dev/null -s -w 'Response Code: %{response_code} Time: %{time_total}s\n' http://localhost:8090/api/mdb/myproject
Response Code: 200 Time: 0.011339s
```
There is no appreciable difference in response times between 1 user and 150 users with Yamcs version 5.7.8

akhenry, 2023-09-26:
We have also observed that as well as long HTTP response times, CPU utilization spikes significantly with authorized requests against a populated user database. This is preventing us from using Yamcs at mission scale right now.

merose, 2023-09-27:
A bit more information:
- Yamcs 5.8.x changed to persist the user database both to a protobuf-style database and to a Yarch database. The Yarch database appears to be the source of the performance problems.
- You can see the performance slowness of the new database by writing a script to add users using the Create User and Update User APIs. When adding 100 users, the performance of each addition decreases. On my laptop, performing a Create User followed by Update User, to add roles, takes about 250ms for the first user but increases dramatically to almost 1500ms by the 100th user. If I comment out the persistence to the Yarch database the problem goes away, and the time for each request is on the order of 25ms.
- The performance slowness also affects HTTP requests using authentication, when HTTP basic authentication is used. When basic authentication is used, every request requires recreating the system and object permissions for a user, followed by a persist of the user database, causing every single HTTP request to be delayed by hundreds of milliseconds if there is a reasonable number of configured users.

It appears that there are several problems:
1. Yamcs should not persist the user database to disk unless a user's properties have changed.
2. Yamcs should not persist the user database in the foreground during a request, as slow persistence will slow down each HTTP request.
3. The persistence to the Yamcs database seems extremely slow, an order of magnitude slower than the older, protobuf-based persistence.
4. There is no way to replace the Directory class with a more performance or alternate storage mechanism. There is also no way to avoid using the Directory class even if you are using an external user database, as the HttpRequestHandler always talks to the Directory when using HTTP basic authentication.

We have worked around number 1 above by adding a "dirty" flag to the User object and avoiding persistence if the User object has not been changed. This drops the time for authenticated requests down to the almost the time for unauthenticated requests. However, we had to override both the Directory and User objects to make this change.

merose, 2023-09-27:
Also, I have confirmed the problem exists in Yamcs 5.8.6 using the quickstart project.

fqqb, 2023-09-28:
Thanks, will investigate soon

fqqb, 2023-09-28:
For information, 5.8.4 did include a commit that introduces a user cache of some kind, in response to another issue about a race:
https://github.com/yamcs/yamcs/commit/e4d2ad7093731d679ea5991669e8a864bc9062b7

Not saying that's a full answer to this issue, I will need to investigate further.

fqqb, 2023-10-04:
Using your script, I can indeed confirm undesirable slowdown. The main contributor to that appears to be query compilation, which is happening very often due to the full-rewrite of the accounts table, which are re-inserted one-by-one.

And as per Mark's analysis, this is indeed occuring on every request due to the use of Basic Auth, which goes through the login-process each time.

I'm working on updates to make this all faster.

fqqb, 2023-11-03:
Fixed as of v5.8.8

closed 2023-11-03 by fqqb